horizon-eda/horizon #570: Feature Request: Impedance Matching

trembel, 2021-02-11:
It would be nice to have an integrated impedance-matching tool, similar to what altium does (https://www.youtube.com/watch?v=0Bok7KpZ_aw), s.t. a rule directly can access an impedance profile (which is adjusted if the layer stack is adjusted). For this it would be needed to add the dielectric constants to the layer stackup.

Best,
Trembel

carrotIndustries, 2021-02-11:
I remember having discussed this topic somewhere else already. The thing is that I at least don't trust a single source for impedance calculations and I don't want people to blame Horizon EDA for getting their controlled impedance lines wrong.

Apart from that, I think that this feature doesn't have a particular good effort to incremental value ratio. All it'll do is save the user from copy/pasting a couple of numbers from an external impedance calculator into the rules.

RX14, 2021-04-09:
I made the oriignal request for this, but I've been convinced by multiple sources that the only correct way to calculate impedances is with a field solver, and all the existing formulas, including the IPC methods, work only over a few small ranges of values.

carrotIndustries, 2021-06-06:
> Suggest this is closed.

closed 2021-06-06 by carrotIndustries