horizon-eda/horizon #571: Flags for parts

carrotIndustries, 2021-02-14:
https://github.com/horizon-eda/horizon-pool/pull/193 @jue89 got me thinking that we finally need to add flags to parts to store details such as that particular part not being an actual part that one can buy and place on a board. Implementing them is easy, the hard part is figuring out which flags we need.

Some ideas:
 - Exclude from BOM
 - Exclude form pick&place export (can this be merged with the one above?)
 - Base part

jue89, 2021-02-20:
> Some ideas:
> 
> * Exclude from BOM
> * Exclude form pick&place export (can this be merged with the one above?)
> * Base part

I think those three flags are at least a good starter. After having thought about this during the last days, I don't see a relevant flag missing in your list.

carrotIndustries, 2021-04-17:
The mentioned commit adds the mentioned flags. Right now, the base part flag doesn't do anything. The most sensible thing to do with it would be adding an option to the part browser that excludes base parts. Any other ideas?

For enhanced compatibility with older versions, the version number written to the file depends on whether there are any flags set, so users of older releases won't get warnings all over the place if a part was edited with a newer release.

The only drawback is that the we're now stuck with the "This Part has been created with an older version..." bar if a part doesn't have any flags.

RX14, 2021-04-17:
Looks good, adding that filtering (on by default?) to the pool manager looks like the next step.

> The only drawback is that the we're now stuck with the "This Part has been created with an older version..." bar if a part doesn't have any flags.

Hopefully this would be only for a release or two until users of older versions would be expected to upgrade to use the latest pool commits.

carrotIndustries, 2021-04-17:
> Hopefully this would be only for a release or two until users of older versions would be expected to upgrade to use the latest pool commits.

Unfortunately that's not the case: That warning appears whenever the app version is greater than the file version. The app version got bumped to 1 for the flags.

When saving a part and there aren't any flags set, the file version stays at 0 even if the app version is 1. Only when setting flags, the part is saved with the file version set to 1.

Maybe reword the bar to say "Saving ~will~ might update it to the latest version…"?

> Looks good, adding that filtering (on by default?) to the pool manager looks like the next step.

Just landed on this branch. In the part browser in projects, it's set to off by default.

RX14, 2021-04-17:
> When saving a part and there aren't any flags set, the file version stays at 0 even if the app version is 1. Only when setting flags, the part is saved with the file version set to 1.

I meant you could remove this logic after a version or two. Minor though.

commit d3b3b704 (2021-04-17): add part flags

closes #571
commit e2c14429 (2021-04-17): add part flags

closes #571
closed 2021-04-17 by carrotIndustries