horizon-eda/horizon #572: Edit pad in package editor throws when used on pad from another pool

jmgao, 2021-02-15:
When editing a package in a pool that includes the default pool, double clicking on many of the default padstacks [1] throws with the following backtrace:

```
#0  0x00007f6ae209fbb0 in __cxa_throw () from /nix/store/sipmc4wnbcws4vahqlf5i06zz7xgnp23-gcc-10.2.0-lib/lib/libstdc++.so.6
#1  0x00007f6ae2097133 in ?? () from /nix/store/sipmc4wnbcws4vahqlf5i06zz7xgnp23-gcc-10.2.0-lib/lib/libstdc++.so.6
#2  0x00000000008730c8 in std::vector<horizon::TriangleInfo, std::allocator<horizon::TriangleInfo> >::_M_range_check (__n=<optimized out>, this=<optimized out>)
    at /nix/store/9rmf9whh2m8fqlqp859d8pmsb63jarsp-gcc-10.2.0/include/c++/10.2.0/bits/stl_vector.h:1073
#3  std::vector<horizon::TriangleInfo, std::allocator<horizon::TriangleInfo> >::at (__n=<optimized out>, this=<optimized out>)
    at /nix/store/9rmf9whh2m8fqlqp859d8pmsb63jarsp-gcc-10.2.0/include/c++/10.2.0/bits/stl_vector.h:1094
#4  horizon::vector_pair<horizon::Triangle, horizon::TriangleInfo>::atm (idx=<optimized out>, this=<optimized out>) at src/util/vector_pair.hpp:117
#5  horizon::Canvas::set_flags (this=0x2127860, r=..., mask_set=2 '\002', mask_clear=<optimized out>) at src/canvas/canvas.cpp:119
#6  0x0000000000822429 in horizon::ImpPackage::update_highlights (this=0x1d38520) at src/imp/imp_package.cpp:356
#7  0x00000000009310c0 in horizon::ToolEditPadParameterSet::begin (this=0x52cb700, args=...) at src/core/tools/tool_edit_pad_parameter_set.cpp:35
#8  0x00000000008c22eb in horizon::Core::tool_begin (this=<optimized out>, tool_id=<optimized out>, tool_id@entry=horizon::ToolID::EDIT_PAD_PARAMETER_SET, args=...,
    imp=<optimized out>, transient=transient@entry=false) at src/core/core.cpp:56
#9  0x00000000007d1e30 in horizon::ImpBase::tool_begin (this=0x1d38520, id=horizon::ToolID::EDIT_PAD_PARAMETER_SET, override_selection=override_selection@entry=false,
    sel=std::set with 0 elements, data=std::unique_ptr<horizon::ToolData> = {...}) at src/imp/imp.cpp:1017
#10 0x00000000007f2daa in horizon::ImpBase::handle_tool_action (this=<optimized out>, conn=...) at src/imp/imp_action.cpp:231
```

1: List of the padstacks that do this:
circular mechanical
obround mechanical
smd rectangular (no paste)
smd rectangular (no paste, bottom)
TH obround
TH round
TH round with rectangular pad

carrotIndustries, 2021-02-15:
I can't reproduce this and can't really think of a way why including pools should have an effect on this piece of code. Which version are you using?

jmgao, 2021-02-15:
Whoops, looks like it has nothing to do with it being a separate pool: it's related to the package itself.

Here's an example package that repros for me on tip of tree: https://github.com/jmgao/test_pool/blob/master/packages/test_package/package.json

commit bce19221 (2021-02-16): Canvas: don't crash if rendering empty group

fixes #572
closed 2021-02-16 by carrotIndustries