horizon-eda/horizon #576: Generic parts

bob-u, 2021-02-22:
Whenever I come back to check on the progress I just cannot go past that:  Why is Horizon EDA forcing me to chose manufacturer or part number as the first step of designing a circuit?   Especially since not all part variations are in the database.

For discrete parts it would be great if I could just pick part's parameters that are necessary for functioning of the circuit.  Foot-print, value and tolerance.  It would be nice if tolerance was a range.

RX14, 2021-04-05:
What I do in this case, is to pick "base <package> capacitor/resistor/inductor" as the part, and set the correct value as the custom value. There should probably be more discussion on this, but the workflow isn't actually that much worse than other EDA tools, just less immediately obvious.

frmdstryr, 2021-04-21:
I was thinking the other day, having a notes section for each part placed would be nice.