horizon-eda/horizon #578: Feature request: Board rules -> regex for net class names

jue89, 2021-03-03:
I've created a board with 3 different galvanically isolated zones. Each zone has two own net classes for power and signal - resulting in 6 net classes:
- A_signal
- A_power
- B_signal
- B_power
- C_signal
- C_power

A regex match for net classes would allow to formulate these rules:
- `signal$`: default track width 150um
- `power$`: default track width 500um
- `^A` -> `^[^A]`: min. copper clearance 5mm
- `^B` -> `^[^B]`: min. copper clearance 5mm
- `^C` -> `^[^C]`: min. copper clearance 5mm

The last three rules are not very readable - to be honest. Maybe the 2nd Match could get another option that matches every other entity not matching the 1st Match?

carrotIndustries, 2021-03-06:
see the linked branch for the requested feature

jue89, 2021-03-07:
Wow! Really nice! Just tried out my described use case and it looks great :)

[image]

commit f2e83ac5 (2021-03-07): rules: add net class regex matching

closes #578
closed 2021-03-07 by carrotIndustries