horizon-eda/horizon #581: Zoom by keyboard center point does not follow mouse

MarSik, 2021-03-08:
When I use the Zoom keystrokes (Numpad +/-) the screen stays centered as is. I would prefer if the zoom center point followed the mouse.

(KiCAD behaves like that for what it is worth + it always moves the zoom center to the center of the screen).

carrotIndustries, 2021-03-08:
Dunno, I implemented it that way, since people might have the mouse cursor at some random location when they zoom using the keyboard shortcuts. If they were actively using the mouse, I'd expect them to zoom using the scroll wheel.

What's your use case?

matt-roberts, 2021-03-15:
I came here to log this issue as well.  Anyway my use case is as follows:

In an effort to avoid pain in my hand; I have removed the mouse wheel (and therefore also the middle button).  This means that I can use the keypad + / - keys to zoom all the time, and would like them to zoom on the mouse cursor.

Incidentally, if you are laying a trace in the board view the end of the trace follows the mouse pointer (which is good), but zooming with the keypad keys seems to be based on the center of the view.  the consequence is that if you tap keypad + / - while laying out a trace it is really confusing.

MarSik, 2021-03-15:
My use case is a laptop when I am on the road. I use trackpoint and only have two reliable mouse buttons and no scroll wheel. (And no numeric block, but I can fix the hot keys easily).

carrotIndustries, 2021-03-15:
added an option for this in 89c3e67fc268e9a333b9a505be2d0fb46dd57bb9

[image]

closed 2021-03-15 by carrotIndustries

matt-roberts, 2021-03-16:
Fantastic.  Thank you.