horizon-eda/horizon #582: Option to disable dragging components

RX14, 2021-03-12:
I find often I end up accidentally dragging components when I'm trying to drag a box selection. I'd prefer to have the option to never drag components, like there's already an option to disable dragging to start tracks.

carrotIndustries, 2021-03-12:
Should this be a global setting that applies to dragging in general in all editors?

RX14, 2021-03-13:
yes please

closed 2021-03-14 by carrotIndustries

RX14, 2021-03-14:
I think i discovered *why* I hated it so much though, the hover-select mode clashes with drag on move because it lets you drag any item without having to click on it first. A good default would be to disable drag to move always when you're in hover select mode, and retain the option to disable it always.

commit 3ac2d4e3 (2021-03-14): add option for disabling drag-to-move

closes #582