horizon-eda/horizon #584: consistency problems that I have noticed

matt-roberts, 2021-03-15:
In the 3D preview... shift-left-drag works the same as in the board editor.  But keypad + / - do not zoom like they do in the board editor.

When editing a plane priority in the right panel in the board view, return/enter doesn't finishing the number entering.  (as a result I have confused look on my face and a priority like "10q".  The intention was to set the plane priority to 10, press enter, and then press q to update all the planes to see if I have chosen the correct number).

In both the schematic editor and the board editor: you can press ctrl-arrow to move the view.  but that view only updates when you move the mouse.

carrotIndustries, 2021-03-15:
> In both the schematic editor and the board editor: you can press ctrl-arrow to move the view. but that view only updates when you move the mouse.

Which version are you using? This should have been fixed by https://github.com/horizon-eda/horizon/commit/edfe7ed75de62c20f83597c671b7a15fd6391978

matt-roberts, 2021-03-15:
I am using v1.4.0 on windows, downloaded from https://github.com/horizon-eda/horizon/releases/download/v1.4.0/horizon-eda-1.4.0-x64.msi

So I guess I have that fix to look forward to in the next compiled release.

carrotIndustries, 2021-03-15:
> So I guess I have that fix to look forward to in the next compiled release.

I'd be glad if you could try the latest zip from https://ci.appveyor.com/project/carrotIndustries/horizon/builds/38224063/artifacts to make sure that this is actually fixed. I recommend creating a copy of projects before opening them in that build as it will migrate projects to the new project pool.

matt-roberts, 2021-03-15:
Yep, Ctrl+Arrow works as expected with that version on my end.

carrotIndustries, 2021-03-16:
> When editing a plane priority in the right panel in the board view, return/enter doesn't finishing the number entering

fixed in 31f296de5a913d030671de5d4a53b27f3960159a

> Yep, Ctrl+Arrow works as expected with that version on my end.

Thanks for testing!

carrotIndustries, 2021-03-16:
> But keypad + / - do not zoom like they do in the board editor.

That's a bit more complicated that it may seem on the surface: In the board/schematic/etc.. editors, these are implemented as actions that can be assigned arbitrary key bindings. The proper way to implement this would be to add actions to the 3D preview as well. That way, the 3D preview can use the same keybinding that are defined for the other editors.

matt-roberts, 2021-03-17:
Oh I assumed it was closer to trivial than that.

In the meantime, on windows, it is possible to bodge it using AutoHotKey.  If anyone is curious here is a script that does it:

```
#If WinActive("ahk_exe horizon-imp.exe") and WinActive("3D Preview")
NumpadAdd::
  MouseClick,WheelUp,,,1,0,D,R
return

#If WinActive("ahk_exe horizon-imp.exe") and WinActive("3D Preview")
NumpadSub::
  MouseClick,WheelDown,,,1,0,D,R
return
```

commit 54d3eb17 (2021-06-19): 3d preview: support action keys

closes #584
closed 2021-06-19 by carrotIndustries