horizon-eda/horizon #585: Feature Request - View an Enter exact coordinates

matt-roberts, 2021-03-15:
The ability to see the X,Y coordinates of things in the right panel, and possibly a key to edit them:

- this already exists for many things (like part locations in the board editor)
- but when I wish to see where the corner of a polygon (say a copper plane) is, I have to use the "Enter Datum" feature to see where it is.
- and nothing seems to let me see or edit the exact position of vias (I'd like to have a row of say 50 vias have equal spacing and be a line).
- similarly, I'd also like to be able to the same with symbol and junction placement in the schematic.  (less of an issue for me as I've been using the "m" move command and "/" to make it either change just one coordinate to line things up).

carrotIndustries, 2021-03-16:
> but when I wish to see where the corner of a polygon (say a copper plane) is, I have to use the "Enter Datum" feature to see where it is.

This is somewhat more involved, as polygon vertices are referenced by UUID and vertex, whereas other things like packages only require a UUID for identification. Everything in the right panel and the supporting infrastructure is built around an object being identifiable by UUID only. For now, the substantial increase in complexity didn't seem worth it.

1ab69c4e2bc916efe299ecde9fd1f960a4d0d99e now shows polygon vertex positions in the HUD

> and nothing seems to let me see or edit the exact position of vias

done in 4f00674dd68cc48907828f84600c6571eadb9a18

> similarly, I'd also like to be able to the same with symbol and junction placement in the schematic.

I don't really think that this use case is worth the extra space in the right panel. Also, we'd have to make sure that everything stays on-grid.

matt-roberts, 2021-03-17:
> I don't really think that this use case is worth the extra space in the right panel. Also, we'd have to make sure that everything stays on-grid.

No problem, as I said it can we worked around reasonably at the moment.

frmdstryr, 2021-05-20:
> and nothing seems to let me see or edit the exact position of vias 

Was about to open an issue to request this but see it's already done. Nice!