horizon-eda/horizon #588: 3D models are not cached correctly

fruchti, 2021-03-27:
Steps to reproduce:

- Check out a pool commit with an extra 3D model (I used [this PR](https://github.com/horizon-eda/horizon-pool/pull/133))
- Open a project referencing this pool and place the new part in the schematic
- Check out another pool commit without the part’s 3D model (e.g. master) and update the pool from the project pool window
- Place the part’s package on the project’s board (works as the package is cached)
- Try to open the 3D view, which fails with an error message:

[image]

In my test project, the 3D model wasn’t copied to the project pool directory. The message also indicates the cached package references the 3D model path within the main pool.

carrotIndustries, 2021-03-27:
3D models are lazily copied when loaded, so the model not being copied is to be expected.

> The message also indicates the cached package references the 3D model path within the main pool.

This message appears since it tried copying the 3D model from the main pool and the 3D model didn't exist.

closed 2021-03-27 by carrotIndustries

fruchti, 2021-03-27:
Ah, I see. Thanks for the very quick fix!

I quite often check out branches of the main pool, add parts to a project and rely on the cache after that. Opening the 3D view to make sure the model is copied will be an extra step for me, but I don’t have a better solution without dropping the lazy copy either.

commit 4f27b5f7 (2021-03-27): ProjectPool::get_model_filename: don't crash if src isn't found

fixes #588