horizon-eda/horizon #590: Options to derive project/schematic date and revision from git

RX14, 2021-04-09:
The schematic frames and project info allows you to enter a date and revision, which are tricky to keep in-date. It'd be nice to have the option to derive this from git, probably using tags for the revision and last commit or tag date for the date. This would ideally be customizable.

carrotIndustries, 2021-04-18:
When and where should this happen?

RX14, 2021-04-18:
Schematic properties -> global is where you edit the date, presumably that'd be where you set things up. In terms of updating the info, that's a little trickier, though definitely when doing PDF export/printing schematics.

carrotIndustries, 2021-06-06:
> though definitely when doing PDF export/printing schematics.

That has some rather unexpected side effects. We want this to be part of the undo/redo history for consistency reasons. Modifying the schematic/board during pdf export likely isn't what people expect.

To open the git repository, we'd also need to know where on the filesystem the project is and the schematic properties dialog by being a tool doesn't know anything about filenames.