horizon-eda/horizon #592: Different grids in different parts of the board

RX14, 2021-04-28:
Different parts come with different pin pitches, and may even have exact mechanical location requirements which aren't on-grid. It's nice to have the fanout from components be on a grid with a pitch the same as the component itself.

Altium seems to solve this with the ability to have different grids in different parts of the board. This is a fairly complex feature but I'm wondering if this was something you'd be interested in implementing or if you have other ideas about grids.

carrotIndustries, 2021-04-28:
I think that having multiple grids would complicate things too much right now. After all, maintaining Horizon EDA is a constant fight against things becoming too complex. While some things are easy to tack on and don't interfere with much else, other things have a habit of popcorning in complexity. 

A rather easy to implement substitute could be adding a way of saving and recalling grid spacing and offset, so one could quickly recall the grid needed for the job. There's nothing preventing this from happing when a tool is active, so it'd be possible to start routing with one grid and switch to another one midway.

RX14, 2021-04-28:
Yeah, I think that's a really neat compromise! I'd be on board with that.

carrotIndustries, 2021-06-06:
See the linked branch. 
[image]

I'm not 100% sure about the layout of the window, it looks a bit uninspired to me. Also, activating an item in the list keeps the window open. Do you think a "select grid" action that brings up a different window that closes on activating a grid makes sense?

RX14, 2021-06-06:
> Do you think a "select grid" action that brings up a different window that closes on activating a grid makes sense?

yeah that sounds okay, or making it a part of the already-existing grid configuration collapsible in the left sidebar

carrotIndustries, 2021-06-07:
> yeah that sounds okay, or making it a part of the already-existing grid configuration collapsible in the left sidebar

We now have that select grid action. It's actually the same window as the grids window, but I don't see how this could be an issue.

RX14, 2021-06-08:
Perhaps:
- "Save": create a new entry with the current grid settings
- "Load": load entry into current grid settings
- "Remove": add confirm dialog and delete current grid
- the current ""load"" IMO isn't that useful and confuses the terminology

RX14, 2021-06-08:
Plus the grids should be saved into the project file and not imp meta IMO, they're very useful to share across collaborators.

carrotIndustries, 2021-06-08:
> "Save": create a new entry with the current grid settings

I'm actually fine with create (or new) as it adds a new item to the list.

> "Remove": add confirm dialog and delete current grid

Do we really need a confirmation dialog? Accidentally deleting a grid setting shouldn't be that big a deal.

> the current ""load"" IMO isn't that useful and confuses the terminology

Without it, one would have to delete an re-add a grid to modify it. Seems somewhat cumbersome to me.

> Plus the grids should be saved into the project file and not imp meta IMO

Storing things like that in the board file itself was always a bit hacky as it involved re-reading the board file and monkey-patching the json document when saving it.

Right now, we store these things in the imp_meta file:

 - Display options tab
 - Selections made in the "nets" window
 - Parts selected in the "parts" window
 - Layer settings
 - Grid settings
 - Grids as defined in the grids window

Do you think that anything else other than the last one is worth sharing across users? Introducing yet another way of where to store things isn't really worth it for just that one thing, as far as I'm concerned. Any suggestions for names?

RX14, 2021-06-08:
I won't argue the naming cause it's just personal taste. Though I don't believe that editing a grid will be a common operation.

> Storing things like that in the board file itself was always a bit hacky as it involved re-reading the board file and monkey-patching the json document when saving it.

This seems a little strange...

RX14, 2021-06-10:
IMO the grids you use are just as important as the rules you set up...

carrotIndustries, 2021-06-19:
> IMO the grids you use are just as important as the rules you set up... Surely they could be stored the same way.

Having though about this for a bit, I'm convinced that it's a good idea to store them in the board file same as rules and export settings.

Should we keep the grid settings for packages in `imp_meta` or move them into the `package.json` file?

RX14, 2021-06-19:
I don't mind too much, but I think it's nicer in package.json, i've noticed this before.

carrotIndustries, 2021-06-20:
See the branch for an updated version that stores the grid settings in board/package.

RX14, 2021-06-20:
- Editing the grid settings and nothing else doesn't allow you to save.
- Every time I use the dialog, I accidentally press "load" instead of "apply" and then there's no ctrl-z way back. If these names stick, there needs to be confirm dialogs on destructive actions, or ctrl-z.
- I still really dislike the "load" name too, to my brain it signifies "load the currently selected grid (to the board)" and not what it curently does, which is the opposite.

carrotIndustries, 2021-06-20:
>    Editing the grid settings and nothing else doesn't allow you to save.

Fixed 

>    Every time I use the dialog, I accidentally press "load" instead of "apply" and then there's no ctrl-z way back. If these names stick, there needs to be confirm dialogs on destructive actions, or ctrl-z.
 >   I still really dislike the "load" name too, to my brain it signifies "load the currently selected grid (to the board)" and not what it curently does, which is the opposite.

Since I couldn't come up with a better name that button is now gone.

As we're now down to three buttons, we could replace the them with icons: plus (for add), minus (for remove) and a tickmark to apply the selected grid.

RX14, 2021-06-21:
Thanks!

> As we're now down to three buttons, we could replace the them with icons: plus (for add), minus (for remove) and a tickmark to apply the selected grid.

I think that'd look good.

RX14, 2021-06-21:
okay i tried it, it does look good, I think it's ready to merge

commit 576654bd (2021-06-20): imp: add grids window

closes #592
commit 52fb1fc5 (2021-06-21): imp: add grids window

also move grid settings into board/package

closes #592
closed 2021-06-21 by carrotIndustries