horizon-eda/horizon #593: Board-level keepouts not working

bsilver8192, 2021-05-02:
I can't get a board-level keepout to render in the gerber output. It's possible I'm not selecting the necessary buttons, but I think I've got it.

I'm doing this (with the top copper layer selected):
1. Draw a plane on the top copper layer
2. `Draw keepout`, draw a shape contained within the plane
3. `Fabrication output`, no evidence of the keepout

Note that package-level keepouts are working fine for me, which is a tolerable workaround for what I'm doing. I'm using 1.4.0.

carrotIndustries, 2021-05-02:
It's working as intended, why should keepouts show up in the gerber output? Their purpose is to denote the absence of copper.

bsilver8192, 2021-05-02:
I meant there should be a hole in the plane in the gerber output where the keepout is, but there isn't. But now as I try doing creating a plane from scratch, it works. I think I've got my plane in a weird state where it's filled but has no pins in it. I'm not sure how I created that, when I try doing it again I can't get it to fill in without putting a pin somewhere. Any ideas on what caused that? Possibly also related: this plane is imported from a DXF and has arcs.

I'll send a copy of the project in a bit, after cleaning it up.

carrotIndustries, 2021-05-02:
Did you run "update all planes" after placing the keepout?

bsilver8192, 2021-05-02:
Sorry for the noise, apparently I just needed the prompt to look through everything again. Works great now. Thanks for the responsiveness!

For the curious: I had converted my plane converted back to a polygon while trying to make it fill the plane without adding a pin, and then forgot about it. Keepouts don't work with polygons... I got it to fill the plane like I want by setting `Keep orphans` to on, and now the keepouts work too.

closed 2021-05-02 by bsilver8192