horizon-eda/horizon #595: Build fail due to error: filesystem: No such file or directory

fk0815, 2021-05-12:
Building on opensuse 15.2 and also on 15.3 fails since a few days.

build logs:
https://build.opensuse.org/package/live_build_log/home:frank_kunz/horizon-next/openSUSE_Leap_15.2/x86_64
https://build.opensuse.org/package/live_build_log/home:frank_kunz/horizon-next/openSUSE_Leap_15.3/x86_64

both use gcc7. Is there a newer gcc/g++ needed? What is the oldest version that is supported?

carrotIndustries, 2021-05-12:
> What is the oldest version that is supported?

gcc 8.

fk0815, 2021-05-13:
It would be good to add that requirement to the build instructions since the build fails just with a missing header file, which might be confusing then.

carrotIndustries, 2021-05-13:
Done: https://github.com/horizon-eda/horizon-docs/commit/24dfb8ce7289315e9598e49c8932b52dac339910

closed 2021-05-13 by carrotIndustries