horizon-eda/horizon #598: Feature Request - Align / distribute tools

frmdstryr, 2021-05-20:
At times wish there were "alignment tools" similar to what inkscape has.  When mixing different types of components that have different sizes it's hard to get them in a straight line with equal spacing.  

[image]

carrotIndustries, 2021-05-20:
Good to know, I've thought of this a as well. The only thing I haven't quite up my mind about yet is how to deal with things such as "align left" as the bounding box of packages is fairly arbitrary.

We also need to figure out for which object types this makes sense:

JUNCTION: Yes, for aligning vias
BOARD_HOLE: Yes
TRACK: Can't think of any use for that since a track is defined by its from and to junction
POLYGON_EDGE,POLYGON_VERTEX,POLYGON_ARC_CENTER: Should this move the polygon as a whole?
TEXT: Yes
LINE: See track
ARC: Makes even less sense than lines
BOARD_PACKAGE: Yes, of course
VIA: Attached to a junction
DIMENSION: Can't think of any use case
PICTURE: Yes
BOARD_PANEL: Yes
PAD: Yes

frmdstryr, 2021-05-20:
List seems right.

> TRACK: Can't think of any use for that since a track is defined by its from and to junction

I don't think aligning tracks makes sense, but if they could be evenly distributed it would be amazing. I've killed so much time adjusting the spacing of adjacent tracks so it "looks" equal...

> Should this move the polygon as a whole?

Played with inkscape a little and it seems like they don't let you align vertices or edges, only the whole thing.

bob-u, 2021-08-31:
I just posted similar question on discourse.  Now that JLCPCb and PCBWay do Aluminum PCBs making your own LED boards is cost effective, just need a way to arrange the pads.

What I'm looking for is ability to distribute LEDs in patterns, rectangular grid, circular with all pads horizontally aligned, circular with pads aligned to circle radius.

carrotIndustries, 2021-09-19:
Here's what it looks like: 
[image]

Hovering over the buttons updates the canvas to preview the result of the operation.

closed 2021-09-19 by carrotIndustries

bob-u, 2021-09-20:
Thank you for implementing it.  I tried to test however compiling fails.  

Also, have you considered circular distribution?

carrotIndustries, 2021-09-20:
> tried to test however compiling fails.

Not exactly helpful. Which OS/what's the error message you're getting? There have been some after-the-fact fixes to make it built on some distros: https://github.com/horizon-eda/horizon/actions

> Also, have you considered circular distribution?

That'd be part of a future "Arrange" tool.

bob-u, 2021-09-21:
I had same errors as in here: https://github.com/horizon-eda/horizon/runs/3644423531?check_suite_focus=true
I'll try to get more info when I have access to that PC again.  It's running MX Linux 19 which successfully compiled Sept 8 commit before.

carrotIndustries, 2021-09-21:
> I had same errors as in here

Fixed in latest master.

bob-u, 2021-09-21:
Got it compiled today, no problem.  It works well and is self explanatory thanks to instant preview.

I would like if there was a way to control space between each object.  Would that be a feature of Arrange?

commit 30615c80 (2021-09-19): new tool: align & distribute

closes #598