horizon-eda/horizon #599: Update planes ignores polygons

frmdstryr, 2021-05-20:
I'm trying to smoothly join two different track widths with a polygon but probably doing it wrong...

[image]

When I update the ground plane it seems to ignore the polygons and overlaps the polygons I added.

[image]

frmdstryr, 2021-05-20:
As suggsted on IRC, I tried selecting it and using "Add plane" then setting the net. But it just doesn't fill it.

[image]

After update all planes

[image]

frmdstryr, 2021-05-20:
Decided to just draw a bunch of overlapping tracks...

closed 2021-05-20 by frmdstryr

carrotIndustries, 2021-05-20:
You might try keeping orphans enabled to prevent it from deleting the fragment that isn't connected to anything.

Thinking a bit further, I think there could be value in properly supporting non-plane polygons on copper layers with a net assigned to them, for example for making RF structures or things as in this issue.