horizon-eda/horizon #601: build failure for python module on opensuse tumbleweed

fk0815, 2021-06-02:
build of python module fails with new release 2.0.0

Build recipe: https://build.opensuse.org/package/view_file/home:frank_kunz/python-horizon/python-horizon.spec?expand=1

[ 482s] build/picobj/src/canvas3d/canvas_mesh.o
[ 482s] build/picobj/src/canvas3d/wall_renderer.o
[ 482s] In file included from src/canvas3d/wall_renderer.cpp:1:
[ 482s] src/canvas3d/wall_renderer.hpp:15:29: error: 'size_t' was not declared in this scope; did you mean 'std::size_t'?
[ 482s] 15 | std::unordered_map<int, size_t> layer_offsets;
[ 482s] | ^~~~~~
[ 482s] | std::size_t
[ 482s] In file included from /usr/include/c++/11/type_traits:38,
[ 482s] from /usr/include/c++/11/unordered_map:38,
[ 482s] from src/canvas3d/wall_renderer.hpp:3,
[ 482s] from src/canvas3d/wall_renderer.cpp:1:
[ 482s] /usr/include/c++/11/x86_64-suse-linux/bits/c++config.h:2430:33: note: 'std::size_t' declared here
[ 482s] 2430 | typedef SIZE_TYPE size_t;
[ 482s] | ^~~~~~
[ 482s] In file included from src/canvas3d/wall_renderer.cpp:1:
[ 482s] src/canvas3d/wall_renderer.hpp:15:35: error: template argument 2 is invalid
[ 482s] 15 | std::unordered_map<int, size_t> layer_offsets;
[ 482s] | ^
[ 482s] src/canvas3d/wall_renderer.hpp:15:35: error: template argument 5 is invalid
[ 482s] src/canvas3d/wall_renderer.hpp:16:5: error: 'size_t' does not name a type
[ 482s] 16 | size_t n_vertices = 0;
[ 482s] | ^~~~~~
[ 482s] src/canvas3d/wall_renderer.hpp:4:1: note: 'size_t' is defined in header ''; did you forget to '#include '?
[ 482s] 3 | #include <unordered_map>
[ 482s] +++ |+#include
[ 482s] 4 |
[ 482s] src/canvas3d/wall_renderer.cpp: In member function 'void horizon::WallRenderer::push()':
[ 482s] src/canvas3d/wall_renderer.cpp:69:5: error: 'n_vertices' was not declared in this scope
[ 482s] 69 | n_vertices = 0;
[ 482s] | ^~~~~~~~~~
[ 482s] src/canvas3d/wall_renderer.cpp:76:19: error: request for member 'clear' in '((horizon::WallRenderer*)this)->horizon::WallRenderer::layer_offsets', which is of non-class type 'int'
[ 482s] 76 | layer_offsets.clear();
[ 482s] | ^~~~~

the full log can be found here: https://build.opensuse.org/build/home:frank_kunz/openSUSE_Tumbleweed/x86_64/python-horizon/_log

carrotIndustries, 2021-06-02:
We now have CI based on arch linux to build with gcc 11. I tried setting up CI based on opensuse, but zypper is surprisingly slow and tumbleweed occasionally breaks: https://github.com/horizon-eda/horizon/runs/2731623931?check_suite_focus=true

closed 2021-06-02 by carrotIndustries

fk0815, 2021-06-03:
I run a nightly build fo opensuse here: https://build.opensuse.org/package/show/home:frank_kunz/horizon-next

commit 3a4e7987 (2021-06-02): fix python module build with gcc11

closes #601