horizon-eda/horizon #602: UX improvements for bus management

jue89, 2021-06-07:
Maybe I'm holding it wrong - but i think, a few improvements could be done to the bus management and handling:

The window for adding a sequence doesn't remember the last entered data. If I'd like to create an bus for 4 buttons and 4 LEDs, it would improve UX if at least first and last index are preserved.

When adding bus rippers, Horizon always starts at the first entry of the bus entries and goes thru the list apathetically (?). I'd change this behavior with the following rules:
1. order by the amount of placed bus rippers for a certain entry starting with the entry with the least amount
2. if two entries have the same amount of bus rippers, order alphabetically

With these rules, placing bus rippers can be *resumed* and you don't have to start at the beginning.

carrotIndustries, 2021-06-07:
> With these rules, placing bus rippers can be resumed and you don't have to start at the beginning.

You can press space to change the current bus ripper.

> order by the amount of placed bus rippers for a certain entry starting with the entry with the least amount
>    if two entries have the same amount of bus rippers, order alphabetically

Hm, that might create confusing ordering for bus members with more than one ripper. How about just considering 'none' and 'any' for determining the order?

carrotIndustries, 2021-06-08:
> if at least first and last index are preserved.

done in 0d43d3a94827ab4261aa407398d18d325d9d733a

jue89, 2021-06-09:
> > if at least first and last index are preserved.
> 
> done in [0d43d3a](https://github.com/horizon-eda/horizon/commit/0d43d3a94827ab4261aa407398d18d325d9d733a)

Nice! Thank you!

> > With these rules, placing bus rippers can be resumed and you don't have to start at the beginning.
> 
> You can press space to change the current bus ripper.

Yes, I know. But pressing the space bar every time I continue placing bus rippers is a little bit annoying. The reason why I opened this issue ;)

> Hm, that might create confusing ordering for bus members with more than one ripper. How about just considering 'none' and 'any' for determining the order?

I think the list shouldn't be ordered by these rules. But starting with the element that has the least amount of placed bus rippers would help in all my use cases.

What do you mean by considering 'none' and 'any'? Tbh, I don't get your idea.

carrotIndustries, 2021-06-10:
See the linked branch. 

> What do you mean by considering 'none' and 'any'?

Any bus member with more than one bus ripper would be considered equivalent in terms of the order.

jue89, 2021-06-10:
Just tried your branch and it exactly fits my way how to use the tool! Too me this feels intuitive to continue placing bus rippers.

What do you think? Does this fit the way you used the bus in your projects?

carrotIndustries, 2021-06-10:
> What do you think? Does this fit the way you used the bus in your projects?

I haven't used  buses in a while, so I think it'll be fine.

commit f75da401 (2021-06-10): tool place bus ripper: sort members by bus ripper count

closes #602
closed 2021-06-10 by carrotIndustries