horizon-eda/horizon #606: Rotating text yields unexpected results

berkowski, 2021-06-16:
Text rotation on PCB layers for the standard 90deg increments (0, 90, 180, 270) is always "left to right" and "bottom to top". 

| rotation | Expected | Actual |
|----------|-----------|---------|
|0             | L2R         | L2R      |
|90           | B2T         | B2T      |
|180         | R2L         | L2R      |
|270         | T2B         | B2T      |

[image]

You can work around this for 90 and 270deg by adding a small offset to the angle, but it appears impossible to display text "upside-down".  Mirroring X only moves the anchor point while mirroring Y doesn't appear to do anything.

[image]

RX14, 2021-06-16:
the current behaviour seems right for schematics, and probably even notes layers on PCBs

however, for actual PCB text then this shouldn't happen

berkowski, 2021-06-16:
I'm OK with the current behavior for "paper-printable" things like schematics or notes layers as there are likely not too many use cases for backwards text there.  The inability to mirror or rotate text "dumbly" on the PCB layers is annoying.  And then having text behave differently in different modes may be even more confusing.

carrotIndustries, 2021-06-16:
Since the text object on schematic and board is the same, the only way I see fixing this would be adding an option to texts to make them rotate as everything else. They'd still be mirrored according to the layer.

berkowski, 2021-06-16:
I think that's a fine solution personally.  At least then the actions on text work more as you would expect.

On a side note I've been very happy so far using Horizon on a few small projects coming from years of KiCAD.  The big reason I used KiCAD was to use version control with all schematics and layouts but that wasn't a silver bullet for keeping all the symbols/footprints in sync.  Horizon extending that to pools and now with pool inheritance feels very good.

carrotIndustries, 2021-06-19:
See the linked branch for my take on this:

[image]

berkowski, 2021-06-19:
My personal preference would be to just have text rotate/mirror as expected without checking some option somewhere.  I think having it otherwise makes it less intuitive.

BUT it's just that, my _personal_ preference.  As long as I can flip text around on silkscreen and copper layers without too much hassle I'm happy.  Thanks for enabling it!

carrotIndustries, 2021-06-19:
> without checking some option somewhere.

Backwards compatibility, we don't want existing designs to look differently in a new version.

commit 76d7cec6 (2021-06-19): texts: add option for upside-down text

closes #606
closed 2021-06-19 by carrotIndustries