horizon-eda/horizon #608: Linux-compatible paths on Windows

jbayless, 2021-06-18:
As a Windows user, I shared my Horizon project for the first time today with a colleague running Linux. He reported that all of the schematic symbols were broken because the Pool paths were encoded with backslashes, presumably because I'm running a Windows system.

(I'm fairly sure that while Windows defaults to backslashes, it actually understands forward-slashes too, for the most part. I've never had any trouble opening forward-slash-encoded symbols with Horizon on Windows).

Anyway, in the interests of compatibility, would it be possible to add a preference for always using Unix-style paths within Horizon project and pool files, even when running on Windows? I'm imagining a feature conceptually similar to git core.autocrlf, where git automatically converts CR+LF line endings to LF line endings. Except for backslashes instead.

carrotIndustries, 2021-06-18:
> he reported that all of the schematic symbols were broken because the Pool paths were encoded with backslashes

Did you also share the pool.db? This one isn't supposed to be shared as it'll be rebuilt automatically.

There are some places throughout the design files that might deserve a closer look, such as the export filenames.

jbayless, 2021-06-18:
Good question. Yes I probably did share it; I'll be sure to add it to my .gitignore from now on. I'll let you know if the problem still persists!

jbayless, 2021-06-18:
Actually, is there a complete list of all of the locally rebuilt files? I may as well make sure they all stay outside of version control.

carrotIndustries, 2021-06-18:
Creating a new project creates a gitignore that's set up the right way, so just use that one.

jbayless, 2021-06-18:
Got it! My bad. I must have deleted that by accident during a project reorganization. Thank you so much for your help!

carrotIndustries, 2021-06-19:
> There are some places throughout the design files that might deserve a closer look, such as the export filenames.

These are already converted to forward slashes on windows. The only place I found where backlash paths get stored in projects is in the patched 3d model paths. 3d297464ebca0fb899f427ad040dfb15bbdda0cd fixes that.

closed 2021-06-19 by carrotIndustries