horizon-eda/horizon #612: Support swapping between dual/single unit entities without re-placing symbols

RX14, 2021-08-22:
A couple of times I've had to switch opamps between single or dual versions (i.e. for power dissipation reasons discovered after drawing the schematic), and I've had to delete the symbols and re-place them, messing up nets, etc.

It'd be nice if it were possible to select 2 single opamp symbols in the schematic editor and change the entity to be a dual opamp, and convert a dual opamp entity into two single opamp entities, etc.

Obviously not limited to opamps, this would be useful for FETs too.

carrotIndustries, 2021-08-22:
From an implementation point of view, it'd be easier if the user placed the new multi-gate part first and then invoked a to-be-implemented "swap component" tool. This had the advantage that it'd also be useful for swapping gates across components. How about this instead?

RX14, 2021-08-22:
so the process would be:

1. place new dual opamp and each unit
2. swap units between already-placed single entity and new dual entity
3. delete both single entities

That makes sense, and the unit swap tool will definitely be useful elsewhere!

carrotIndustries, 2021-08-22:
> so the process would be:

Yes, the only open question is whether this should be part of the existing "swap gates" tool or go into its own tool. Since the new operation doesn't respect swap groups, I think it makes more sense in its own tool.

RX14, 2021-08-22:
Yeah, makes sense. Though I guess it feels like it makes swap groups a bit of an arbitrary limitation, since any limitation on this new tool to only work across different unrelated entities would be arbitrary.

carrotIndustries, 2021-08-22:
See the linked branch for an implementation. For the lack of a better name, the new tool is called "exchange gates". To not render swap groups useless, it only operates across components.

RX14, 2021-08-22:
Works well for me!

commit 6df1b9f9 (2021-09-07): new tool: exchange gates

closes #612
closed 2021-09-07 by carrotIndustries