horizon-eda/horizon #617: Scaling broken on wayland

RX14, 2021-10-16:
I'm using horizon eda https://github.com/horizon-eda/horizon/commit/987724c79949204b095990db0fae6349dabe1ac6, and 2x scaling seems to be broken in wayland. I'm using a ryzen mobile laptop with sway 1.6.1 as the compositor.

Unscaled:
[image]

On a scaling 2 monitor:
[image]

carrotIndustries, 2021-10-16:
It works on X11 (and win32 as far as I'm aware).

What's the return value of `get_scale_factor` there?  https://github.com/horizon-eda/horizon/blob/2d1a7a181a398d30214f573a1b92c03a3b67aeeb/src/canvas/canvas_gl.cpp#L155-L159

unrelentingtech, 2021-12-03:
I haven't seen this issue, running a 2x scale monitor on [Wayfire](https://wayfire.org/) — schematic, board, 3D all look fine.

[image]

(and just checked the current git version, it's still fine)

It's *very* weird for such a bug to only happen on some setups…

RX14, 2021-12-05:
What GPU do you have @unrelentingtech ?

unrelentingtech, 2021-12-05:
Vega 56

```
AMD -- Radeon RX Vega (vega10, LLVM 12.0.1, DRM 3.37, 14.0-CURRENT)
Mesa 22.0.0-devel (git-8abfd58ffa)
```

RX14, 2021-12-06:
> I doubt GPU differences would cause this

GL is surprisingly variable between vendors... Unlikely in this case since we're both on AMD though.

I'm unlikely to debug this again until I get a 2x scaling display again, I'm back on 1x ones now.

RX14, 2022-10-26:
Seems like it was a GTK bug, and is fixed.

closed 2022-10-26 by RX14