horizon-eda/horizon #618: Per-component thermal rules

RX14, 2021-10-19:
Currently thermals are defined per-pour and not per-component, making it hard to customize thermals for specific manufacturing issues on specific components as they arise. Having rules which work on component classes, and the ability to adjust thermals per-component would give more flexibility here.

carrotIndustries, 2021-10-19:
> on component classes

We don't have such thing right now. Did you have something similar to net classes in mind? What else could they be useful for?

Just to write down my thoughts on how such a rule could work like:

Match:
 - Component (component selector and component class)
 - Pad? Technically, we could support this. I think this could be useful for connecting power/RF GND pads to the plane without thermals but using thermals for signal grounds.
 - Net

Settings:
 - Connect style: as defined by plane / solid / custom thermal

carrotIndustries, 2022-02-08:
d2da4b6d4e62c4c6c93a2507f1a42e4e272656f5 adds matching multiple nets : 
[image]

RX14, 2022-02-15:
I'm a bit busy between jobs right now  (I'm finally switching careers from software to electronics!) so I'm unlikely to be able to test this myself

commit 7e880dc0 (2022-02-15): board: add thermal rules

closes #618
closed 2022-02-15 by carrotIndustries