horizon-eda/horizon #619: Feature Request: Restrict 45 degrees for polygon

kamtar, 2021-10-29:
I do like using planes a lot in my designs but it's hard to keep them looking neat without 45-degree snap-on/restriction as in Altium.

I'm thinking about writing it myself but I have yet to go through the codebase so I just wanted to mention it, and ask if I'm not missing something and if you think that it's reasonably easy to implement.

carrotIndustries, 2021-10-29:
That's actually rather straightforward to add. The relevant line is https://github.com/horizon-eda/horizon/blob/5194b3aafc7a63a29d883e4fbc43e2efbceeb5ed/src/core/tools/tool_draw_polygon.cpp#L91 It calculates the position of the vertex being placed based on the cursor position and the last vertex position.

As the coordinate calculation is factored out into the `ToolHelperRestrict` class, it might make sense to add the 45° snap in there. That way the move and draw line tool could also make use of it. I wouldn't have it selectable by the usual <kbd>/</kbd> keybinding though as 4 items are a bit too many to cycle through. Instead, I suggest adding a new in-tool action to select the 45° mode.

kamtar, 2021-10-29:
Thanks for the pointers.
Are you interested in including it in the main branch? Should I bother doing a pull request?

carrotIndustries, 2021-10-29:
>  Are you interested in including it in the main branch? Should I bother doing a pull request?

Yes, I think that's a feature worth having.

kamtar, 2021-10-30:
I went with a default key "q" for now.
It gets toggled between on and off by q, also if it's enabled and you press / for the restrict mode it cancels it and jumps to ARB.

It copies one axis and calculates the other one based on which axis is the longest. I was thinking about creating more modes and having auto, X angle, Y angle but I don't see the use in it yet. 

It jumps to angles based on what's closer to the cursor. Meaning if you are doing a 45-degree line and you move your cursor only on one axis it will jump back on the axis rather soon. Any thoughts on this? I was thinking if I shouldn't implement some hysteresis.

I added it to a polygon and line tool, I was looking at adding it to a move tool too but that looks much more involved so I shall leave it for now.

carrotIndustries, 2021-10-30:
Even if it's not completely done yet, it's best to open a PR.

commit b48a3443 (2021-11-04): add 45 degree restrict to draw line & polygon tools

closes #619
closed 2021-11-04 by carrotIndustries