horizon-eda/horizon #621: std::exception unable to open database file when using a fork of horizon-pool

auxym, 2021-10-31:
This is a fresh install of Horizon (2.1.0) on windows 10.

When selecting the default managed pool option, everything seems OK. However, when I add my own fork of the pool, then doing any operation that invokes the pool (opening a project, creating a new project, opening the pool manager) results a en error dialog with the following text:

```
std::exception: unable to open database file

Looks like this is a bug. Please report  this and describe what you did previously.
```

Please let me know if I can provide more information, eg via a log file or verbose mode.

carrotIndustries, 2021-11-01:
Perhaps that's #614.

Does opening the pool from the home screen (the Open… button on the top left) work? This should create all the required database files.

claui, 2021-11-01:
@carrotIndustries I’m not OP but I just happened to run into the same bug, and your suggested workaround works for me. Thank you!

auxym, 2021-11-02:
Can confirm - opening the pool from the first screen's "open" menu does work correctly. Works for me and it is my understanding (from #614) that the fix is already implemented in development versions, so I'm closing.

closed 2021-11-02 by auxym