horizon-eda/horizon #622: Keep-out Polygon Problem

oldrev, 2021-11-10:
Hello:

First, Horizon EDA is very impressive, I can’t imagine how much effort you all put into developing this bad boy, the rotation function in board editor that kills.

At this very moment, I'm trying to draw a simple MCU board, I found the keep-out polygon seems to have no effect on the poured plane in board editor, but it does the job in 3D preview window, see the following pictures:

[image]

[image]

So, my stupid question is: is this a feature or a bug?

Env: Horizon EDA CI 2021-11-09 on Windows 10.

Cheers!

carrotIndustries, 2021-11-10:
This is probably the same issue as https://github.com/horizon-eda/horizon/issues/313. The problem is that keepouts are drawn identically to planes.

oldrev, 2021-11-11:
> This is probably the same issue as #313. The problem is that keepouts are drawn identically to planes.

Thanks for the quick reply, so the workaround solution for the current version is net color?

carrotIndustries, 2021-11-12:
I'll try to implement drawing keepouts as outline only in the next week or so.

On the long run, we want something like per-type fill patterns to make them easier to distinguish.

oldrev, 2021-11-12:
Sounds great, can't wait for the next release.

commit 4f84530a (2021-11-13): imp board: add display keepouts in outline mode by default

add option for filled display

closes #622
closed 2021-11-13 by carrotIndustries