horizon-eda/horizon #623: Incompatible with opencascade 7.6.0

yurivict, 2021-11-15:
opencascade was updated to 7.6.0 in FreeBSD ports and horizon-eda is now failing:
```
src/import_step/step_importer.cpp:190:61: error: no member named 'Nodes' in 'Poly_Triangulation'
    const TColgp_Array1OfPnt &arrPolyNodes = triangulation->Nodes();
                                             ~~~~~~~~~~~~~ ^
src/import_step/step_importer.cpp:191:64: warning: 'Triangles' is deprecated: Deprecated method, Triangle() should be used instead [-Wdeprecated-declarations]
    const Poly_Array1OfTriangle &arrTriangles = triangulation->Triangles();
                                                               ^
/usr/local/include/OpenCASCADE/Poly_Triangulation.hxx:317:3: note: 'Triangles' has been explicitly marked deprecated here
  Standard_DEPRECATED("Deprecated method, Triangle() should be used instead")
  ^
/usr/local/include/OpenCASCADE/Standard_Macro.hxx:154:54: note: expanded from macro 'Standard_DEPRECATED'
  #define Standard_DEPRECATED(theMsg) __attribute__((deprecated(theMsg)))
                                                     ^
src/import_step/step_importer.cpp:192:65: error: no member named 'Normals' in 'Poly_Triangulation'
    const TShort_Array1OfShortReal &arrNormals = triangulation->Normals();
                                                 ~~~~~~~~~~~~~ ^
1 warning and 2 errors generated.
```

Version: 2.1.0

Full log: http://package19.nyi.freebsd.org/data/122amd64-default-build-as-user/913ccecf2a92/logs/horizon-eda-2.1.0_1.log (IPv6 URL)

carrotIndustries, 2021-11-15:
What do I have to tell a freebsd installation to use that version? I've configured the freebsd CI to use freebsd 13, but it's still installing 7.5: https://cirrus-ci.com/task/6051049195700224

yurivict, 2021-11-15:
opencascade was updated in the ports tree's main branch. But does Cirrus CI use the main branch of the ports tree or the quarterly branch? Quarterly branch is still not updated until Jan 1. I don't know how to see in Cirrus CI what ports tree does it use.

yurivict, 2021-11-15:
On a side note, it's strange that the yml code has ```image_family: freebsd-12-2``` [here](https://github.com/horizon-eda/horizon/blob/master/.cirrus.yml#L2) but it actually runs FreeBSD 13.

yurivict, 2021-11-15:
You should add the line like [this project](https://github.com/newsboat/newsboat/blob/8963a85b7382daeee74043b86b423739a711cde9/.cirrus.yml#L41) does.

Additionally, you might want to update the ```freebsd_instance``` section to include ```FreeBSD 12.2``` and ```FreeBSD 13.0``` like they do.
```FreeBSD 11``` is now obsolete.

carrotIndustries, 2021-11-15:
Until the next release, you can apply the linked commit as a patch in the port.

Right now the CI uses 12.2 and judging from the logs https://cirrus-ci.com/task/4852117564293120?logs=main#L32 this is what gets used.

closed 2021-11-15 by carrotIndustries

yurivict, 2021-11-15:
I'll use the patch, thanks.

And you can update your .cirrus.yml as I suggested above to test with a more relevant ports tree and on all currently relevant FreeBSD releases.

carrotIndustries, 2021-11-16:
> I'll use the patch, thanks.

Hang on a second, this patch breaks with older opencascade releases. I'll figure out a patch that works on all releases.

yurivict, 2021-11-16:
Ok, but for me compatibility with just one latest release is sufficient.

carrotIndustries, 2021-11-16:
The patch linked above works for 7.5 and 7.6. I've tested your suggested CI configuration and it seems to work: https://github.com/horizon-eda/horizon/commit/8209a55d4232ad77da587e365803812121b29763 If this is what you had in mind, I'll merge it into master.

yurivict, 2021-11-16:
> [...] If this is what you had in mind, I'll merge it into master.

Yes, please merge it.

Thanks.

commit 96ba1f31 (2021-11-15): STEPImporter: make it compatible with opencascade 7.6.0

fixes #623
commit 0d801d48 (2021-11-15): STEPImporter: make it compatible with opencascade 7.6.0

fixes #623
commit ace34e40 (2021-11-16): STEPImporter: make it compatible with opencascade 7.6.0

this time, don't break everything else

fixes #623