horizon-eda/horizon #624: Feature Request - "Enter Datum" for an existing arc (rounded vertex) should allow entering arc radius instead of edge length

auxym, 2021-11-16:
Currently, when using the "round off vertex" feature, it is possible to hit "enter" to go into "Enter Datum" mode and set the arc radius by typing it in. However, when editing an existing arc, hitting enter will instead set the edge length.

I believe it would be more useful to be able to modify the arc radius after the fact, than the arc length. However, it may be more complex to implement, eg requiring special handling of "rounded off vertex" instead of a generic arc segment? Open to consider alternatives.

carrotIndustries, 2021-11-16:
This best goes into its own tool since it benefits from begin interactive unlike the enter datum tool.

The hardest part will be calculating the limits for the arc radius and the geometry in general. If you want to implement this, I can give you some pointers to get started.

carrotIndustries, 2021-11-18:
While it's not exactly what you need, with d2158ab7be097398da27b3a916537463f295382c dragging polygon arcs does the right thing.

auxym, 2021-11-18:
I'm open to having a look, even though my C++ (and GTK) knowledge is pretty limited. Geometry doesn't scare me though :)

Could you clarify  what you have in mind as "its own tool?" Something in the right click menu, or right toolbar, or in the space bar menu? And yeah, if you could give me a few quick pointers towards which piece(s) of code I should be looking at, that would be helpful for sure.

carrotIndustries, 2021-11-19:
Good news: There's not much Gtk interaction going on in tools.

> Could you clarify what you have in mind as "its own tool"

Generally speaking, a tool is something that modifies the document either interactively (move, draw track/line/polygon) or non-interactively (delete, enter datum). They all live in `src/core/tools`. https://blog.horizon-eda.org/misc/2019/11/25/tool.html gives a somewhat outdated (there's no `commit()` anymore, instead you return `ToolResponse::commit()`) overview of how to implement one.

Usually, the easiest way to get started with a new tool is copying an existing one that does something similar. For our case, that'd be the "round off vertex" tool in `tool_round_off_vertex.cpp` as it has all the ingredients needed for the new tool.