horizon-eda/horizon #625: Feature Request: "Do Not Pop." for a Whole Sheet or a Block?

oldrev, 2021-11-17:
Hello:

Sometimes a function area on a board may not have to be populated for product variants etc., to do that I have to find and click the "Do Not Pop." option in the sidebar again and again.

So, I thought it's could be much more convenient if this EDA software can set the "Do Not Pop." property of every part in a sheet by doing it at sheet level.

IMHO this idea could be a useful feature in Horizon EDA for other ones, plus it's should be easy to implement relatively.

For the long term, maybe this "Do Not Pop." function deserves its own window in the board editor?

Cheers!

carrotIndustries, 2021-11-17:
You can to that with existing features in two ways:

Select the symbols you don't want to to populate and either
 - select the desired population state in the property editor and click "All" to apply it to all symbols selected
 - or use the "mark do not populate tool". You can find it in the context or spacebar menu.

closed 2021-11-17 by carrotIndustries

oldrev, 2021-11-23:
Thanks, I was wondering about the purpose of the "All"  button...