horizon-eda/horizon #626: Project created on Windows cannot be opened on Linux

auxym, 2021-11-21:
## Repro

1. Create a new project on Windows.

2. Open schematic and add a single resistor (or whatever).

2. Copy files to Linux machine somehow. I reproduced this both from syncing over Nextcloud and from a manual copy over samba.

3. Open project on linux machine and open the schematic.

## Result

The following errors are shown in the log, and the component does not appear in the schematic:

```
Seq | Level   | Domain                  | Message; Detail
1   | Warning | Block                   | couldn't load Component 92c5c2b3-a609-4d01-8cfa-62d6bcb43623; file /home/francis/Nextcloud/tmp/backslash-bug/pool/entities\cache\3ce770af-7be0-4f72-b2af-2c8010911aaf.json not opened (startup)
3   | Warning | Schematic               | couldn't load Symbol 5265bbdb-fd31-4e76-9acb-a35ceeedba1f; file /home/francis/Nextcloud/tmp/backslash-bug/pool/symbols\cache\e2cde69b-9fa7-4332-ba8f-7e5b3ee8ad73.json not opened (startup)
```

## Software

Both machines running Horizon EDA 2.1.0. Windows 10 machine installed using official installer, Linux (manjaro) machine from AUR.

## Hypothesis

I'm thinking this might be linked to the backslashes in the paths? I went to looking for them in project files, ripgrepping through the project files, but couldn't find anything.  Therefore they might be getting added by the code?

carrotIndustries, 2021-11-21:
The `pool.db` and `parametric.db` files in the `pool/` directory aren't supposed to portable across installations or and should be excluded when copying projects. They'll automatically be recreated when the project is opened.

If more users run into this, it might be worth adding a check to prevent using pools from other installations.

closed 2021-11-21 by carrotIndustries

auxym, 2021-11-21:
Ah! Makes sense. I can confirm that removing the .db files does indeed eliminate the issue.

However, IMO, syncing project files using dropbox/syncthing/etc or even network shares (samba) might be a somewhat common use case? Deleting the db files before opening the project is a workaround, but not a great solution, especially since the problem is not very apparent. At the very least, maybe a warning message to the user telling them to remove the db files?

Other more transparent solutions:

1. Preferred IMO: dynamically "canonicalize"  all the paths before attempting to load files. Performance impact would be mostly negligible, no? Are there any other portability issues in the db files?

2. Somehow mark the dbs as being valid for a given platform, and force a rebuild when opening on another platform. Could be a bit fragile?

RX14, 2021-11-21:
The intended way of sharing projects is using VCS, where horizon generates the correct `.gitignore` which ignores the db files. If it's possible to reliably delect this situation it'd be great to handle it though.

carrotIndustries, 2021-11-21:
>  Are there any other portability issues in the db files?
>  and force a rebuild when opening on another platform. 

The database files are specific to each version of Horizon EDA. If you were using different versions on each platform, you'd get a DB rebuild every time you switch installations.

I've though about generating an "installation UUID" on first use that's written into all database files. If that UUID doesn't match the local one, the database files will be regenerated.

When using any syncing solution, I highly recommend excluding all database files. Both nextcloud and dropbox appear to support this.

auxym, 2021-11-21:
Alright, I'm sold, I should probably put my stuff in a git repo anyways.

That said, anything other than a cryptic  log message, implying db corruption or missing files, would be better, IMO. Maybe a dialog big explaining that the db appears to be incompatible with the current installation, rebuild Y/N?

carrotIndustries, 2021-11-21:
> rebuild Y/N?

As rebuilding the databases won't cause any harm, my current plan of action involves rebuilding the DB unconditionally if installation UUIDs don't match. I'll probably also log a warning to inform users that they shouldn't copy around databses in the first place.

carrotIndustries, 2021-11-24:
See the linked commit for an implementation of installation UUIDs. If the UUIDs don't match the database is rebuilt and the user is greeted by this info bar: 
[image]

Any comments on the wording?

The installation UUID is generated randomly on first use. If users really think that they need to share the databases across multiple installations, there's still the option to manually modify the installation UUID stored in `~/.config/horizon/installation_uuid.json`.

auxym, 2021-11-26:
Wording is fine by me, but if I were to nitpick, "Don't share" might sound a bit harsh? Personnally I would use one of the following:

* Sharing .db files between installations is not supported.

* Horizon EDA does not support sharing .db files between installations.

The second uses active wording which I believe might be prefferable.

Thanks for the installation UUID implementation, I do believe it is an improvement to Horizon!