horizon-eda/horizon #627: Lagging board and schematic editor

jue89, 2021-11-23:
Sometimes, the board and schematic editor are lagging. It feels like a frame rate of <10 FPS.

I wasn't able to reproduce the problem and it might be related to my Nvidia graphics card. (We all know Linus Torvalds LOVES Nvidia for good reasons :D) Restarting my PC resolves this problem.

Does anyone have a guide for me how to benchmark code execution and how to trace down to the source of the problem? Maybe I'll be able to create a helpful issue the next time I'm facing this problem ...

carrotIndustries, 2021-11-23:
When this happens, can you check the gtk GLArea demo and see if it lags as well? Are you using the proprietary drivers our nouveau?

jue89, 2021-11-24:
I'm using the proprietary driver with Xorg.

For streaming I'm utilizing NVENC for transcoding and have to use the proprietary driver and Xorg.

jue89, 2022-02-03:
I'm experiencing lagging, again! GLArea looks fine. No lagging in the GTK example.

Currently I'm running version 2.2.0.

Any other idea how to find out what's causing the low frame rate?