horizon-eda/horizon #631: 3D preview display chaos on AMD integrated graphic

oldrev, 2021-12-04:
When I open my project on my new AMD R5700U laptop computer (no independent graphic card), I got a big mess, see the following:

[image]

and everything is ok on my old Intel i7-8700 desktop machine (no independent graphic card too):

[image]

My Horizon version is 2021-11-22 nightly running on Windows 10.

I also did try to open the `x-band-transmitter`, the same result.

By the way, the PCB editor on the AMD platform is laggy, but the schematic editor is smooth.

Hope it helps.

carrotIndustries, 2021-12-04:
> When I open my project on my new AMD R5700U laptop computer (no independent graphic card), I got a big mess, see the following:

The 3D preview uses instancing and per-instance vertex data to efficiently render the 3D models in a minimum number of draw calls. My guess is that there's some bug in the AMD windows drivers or I'm doing something slightly wrong that happens to work everywhere else. I've had other people, also with AMD GPUs on Windows, report the same issue to me.

Unfortunately, without a suitable machine at hand, there's no sensible way for me to debug this.

> By the way, the PCB editor on the AMD platform is laggy, but the schematic editor is smooth.

Also if the board is empty?

oldrev, 2021-12-04:
No, an empty board or a super simple board is smooth.

I found when I first open the `x-band-transmitter` project, it reacts smoothly. Then I press `Q` to pour the planes, it becomes laggy.

I can program in C++ but know nothing about OpenGL or 3D stuff, sorry cannot help further.

unrelentingtech, 2021-12-07:
Maybe a good way to avoid crappy GL drivers on Windows would be to use [ANGLE](https://chromium.googlesource.com/angle/angle/+/main/README.md), which [gtk does support](https://gitlab.gnome.org/GNOME/gtk/-/merge_requests/191/diffs?commit_id=af66faf6048977c6c27e17c95f6c3c47440904d9). Since it's GLES, this would require porting shaders to GLSL ES, which is a bit stricter. I'll look into that.

carrotIndustries, 2021-12-08:
> would require porting shaders to GLSL ES, which is a bit stricter. I'll look into that.

Both the 2D Canvas (board, schematic, etc) and the 3D one make use of geometry shaders. As far as I could tell, ANGLE doesn't support these. Rendering of 3D models also depends on `glDrawElementsInstancedBaseInstance` which isn't available in ES.

unrelentingtech, 2021-12-08:
I think it does support geometry shaders, here's [a test](https://github.com/google/angle/blob/16ce5226e896403d7e849b2d793f2979f0f8decd/src/tests/gl_tests/GeometryShaderTest.cpp) for it.

hmm, `DrawElementsInstancedBaseInstance` is in [EXT_base_instance](https://www.khronos.org/registry/OpenGL/extensions/EXT/EXT_base_instance.txt) but that doesn't seem to be widely implemented. ANGLE has a bunch of instance stuff in [ANGLE_base_vertex_base_instance](https://github.com/google/angle/blob/main/extensions/ANGLE_base_vertex_base_instance.txt) but apparently not that one specifically :/

carrotIndustries, 2021-12-09:
Would supporting OpenGL ES entail maintaining two sets of shaders and rendering implementations, or are they compatible enough so that we can have identical code for GL and GLES?

Another upside of supporting GLES would be that it'd make it possible to run Horizon EDA on something else than x86 since all(?) embedded GPUs only seem to support GLES.

unrelentingtech, 2021-12-09:
Generally two sets of shaders wouldn't be required, typically the diff is just

```diff
@@ -1,4 +1,5 @@
-#version 330
+#version 320 es
+precision mediump float;
```

which can be conditionally prepended in code before compiling.

There will be a lot of things ES complains about like integer literals for floats, but the fixed versions will still work on desktop GL.

> all(?) embedded GPUs only seem to support GLES

All(?) embedded GPU *proprietary drivers*. Mesa {Panfrost/Freedreno/V3D/Etnaviv/…} will happily run desktop GL. Well, not GL 4.2 though! [Panfrost was at 3.1 in January](https://www.collabora.com/news-and-blog/blog/2021/01/13/desktop-opengl-3-1-on-mali-gpus-with-panfrost/) for example.

Also to be pedantic "something else than x86" doesn't necessarily imply "embedded GPU". I have an arm64 box with a Radeon RX 480 on my desk :)

carrotIndustries, 2021-12-09:
> I have an arm64 box with a Radeon RX 480 on my desk :)

Out of curiosity, does Horizon EDA run on that machine?

Popular SBCs such as the Raspberry Pi 4 appear to support GLES 3.2, so I guess this is the target we should aim for if we want to support GLES.

unrelentingtech, 2021-12-11:
> Out of curiosity, does Horizon EDA run on that machine?

[image]

Why wouldn't it :)

oldrev, 2021-12-13:
Wow, the dark theme looks fantastic! I wonder is there any chance to add a GTK theme selector like Inkscape on Windows.

[image]

carrotIndustries, 2021-12-13:
>  wonder is there any chance to add a GTK theme selector like Inkscape on Windows.

b2bfa20b3e6cad837906cf3aea2812e387b861e4 adds a dark theme toggle to the preferences

> Why wouldn't it :)

There's always the chance of something not working out on unusual platforms. The combination for FreeBSD, ARM, a non-embedded GPU and Wayland really takes the cake for the most unusual system I've seen Horizon EDA run on.

oldrev, 2021-12-14:
[image]

Dark theme on Windows looks great, it's not every day that your wish comes true, thanks a lot.

moritz-meier, 2022-01-28:
>  I wonder is there any chance to add a GTK theme selector like Inkscape on Windows.

You can add a GTK theme into the `share/themes` folder of your installation and set the GTK theme on Windows by adding an `settings.ini` file to `~/AppData/Local/gtk-3.0/`:

```
[Settings]
gtk-theme-name = <theme_name>
```

[image]

I am using Qogir; works great.

oldrev, 2022-01-29:
> You can add a GTK theme into the `share/themes` folder of your installation and set the GTK theme on Windows by adding an `settings.ini` file to `~/AppData/Local/gtk-3.0/`:
>
> ```
> [Settings] [...]

Very informative, thanks.

[image]