horizon-eda/horizon #632: PCB-graphic-only parts shoud be ignored in BoM and PnP files

oldrev, 2021-12-05:
Hello:

Currently when I place a PCB antenna in the schematic then I have to mark it as "Do Not Pop" otherwise it will be presented in the generated BoM and PnP file. IMHO, a big "X" on the antenna symbol looks wired.

To fix this issue I suggest we could add more "fabrication attributes" in the part editor like KiCAD:

[image]

Additionally, an option in the PnP file exporting dialog that allows users to choose SMT-only or SMT+PHT would be much nicer.

carrotIndustries, 2021-12-05:
> To fix this issue I suggest we could add more "fabrication attributes" in the part editor like KiCAD:

We already have that in the part editor: 
[image]

> Additionally, an option in the PnP file exporting dialog that allows users to choose SMT-only or SMT+PHT would be much nicer.

Hm, in combination with "one file per side", that'd be 4 filename entries, not sure if that's really worth it.

oldrev, 2021-12-05:
Oops, that embarrassing. Is just me or the GNOME style is a little wired for productive software: every widget is fat then you have to hide some treasures.

oldrev, 2021-12-05:
By the way, the "1mm Copper Test Point" part in the official core pool didn't set those flags.

RX14, 2021-12-06:
The flags are fairly new and not set for some older parts. PRs are welcome to the pool!

closed 2022-08-13 by carrotIndustries