horizon-eda/horizon #636: Cannot open pools: ‘no such table: pools_included’

fruchti, 2021-12-10:
Fresh build from master. Opening any pool results in the error message ‘Error opening pool’/‘no such table: pools_included’. Removing the pools database files results in Horizon running a pool update for a bit, then aborting with the same message.

carrotIndustries, 2021-12-11:
Since 0fb3508207357f1e6a694b1d3b875c724bf25388 the settings tab tries to check if a pool listed as an available pool includes the current pool to avoid circular included. One of the registered pools probably didn't get updated in a while and thus didn't have that table.

The linked commit introduces the concept of "pool usability" so that the settings tab and the pools window can check if a pool is usable before trying to interact with it.

For a pool to be usable, these conditions must be met:
  - pool.db exists
  - database has the correct schema version
  - installation UUID matches

closed 2021-12-11 by carrotIndustries

fruchti, 2021-12-11:
Thanks a lot! Indeed, opening all pools in the pool settings once to update them solved it as well.

commit e98516cb (2021-12-11): PoolSettingsBox: only check pools_included for usable pools

fixes #636