horizon-eda/horizon #637: Crash when creating part in a pool

matt-roberts, 2021-12-18:
Steps to reproduce:
- Open a pool
- Select the "Parts" section
- Click "Create" and choose any Entity and Package (doesn't matter which)
- Change the "Name" field to anything, which enables the "Save As" button
- Click the "Save As" button and choose a filename
- Click "Close"
- The "Save Changes Before closing?" dialog appears (which is unexpected, given that no changes have happened since I just saved)
- Regardless of if you select "Close without Saving" or "Save" horizon-eda will crash

"About" says I am using "2.1.0 Green Flash", "Commit 362d685 2021-09-07 19:16:05 +0200"
I am running on Windows

If you are curious: I found this bug when I paused part way through creating a microcontroller part.  So I decided to try to save my progress so far, but also leave the window open so that I wouldn't forget to come back to it.

carrotIndustries, 2021-12-19:
I can't reproduce this behaviour on linux. Neither with 2.1.0, nor latest `master`. 

> Click the "Save As" button and choose a filename

And save it or click cancel?

> Click "Close"

Closing the part editor?

matt-roberts, 2021-12-19:
> I can't reproduce this behaviour on linux. Neither with 2.1.0, nor latest `master`.
Interesting.  So it might be a windows only bug?

Sorry for poor quality reproduction steps:
> > Click the "Save As" button and choose a filename
> And save it or click cancel?
Yes, save it.

> > Click "Close"
> Closing the part editor?
Correct.  Close the part editor, which brings up the "save changes before closing?" dialog

I'll exercise it a bit more and see if it actually creates a file at any stage.

Do you have a link to the latest master compiled for windows?  There is always the chance that the bug has been fixed incidentally (clutching at straws I know).

matt-roberts, 2021-12-19:
Okay  this is maddening.  Now I cannot reproduce the bug.  Which means, at the very least, there is something odd going on.  But, I have more evidence about what might have been happening:

I don't think it was saving any json files when I clicked "Save As" before, nor if I selected "Save" in the "Close without Saving?" dialog.  What I mean is that those json part files don't exist in the filesystem now.  (and, I'm fairly certain that I did not tidy them up manually).

Today:
- I click "Save As", and choose a filename (and actually save it)

This time, I checked, and the part json file was written to the disk at this time, and the part appears in the pool manager.

- Close the part editor  (previous this step was called click close)

No longer brings up the "Close without Saving?" dialog.  Which suggests that horizon-eda knows that it has saved the part correctly (and that it knew that it didn't save correctly when I was creating the test case yesterday).

carrotIndustries, 2021-12-19:
> Do you have a link to the latest master compiled for windows? 

https://horizoneda.jfrog.io/artifactory/horizon-win64-ci/horizon-2021-12-18-1409.zip they're linked in the actions: https://github.com/horizon-eda/horizon/actions/runs/1596045860

matt-roberts, 2021-12-26:
I think I now know what I actually did, which is subtly different to the set of steps that I documented above:

1. hidden setup steps:
    - create a part
    - save it (if it helps, imagine you are trying to test the steps for a bug report: so you're being a bit lazy with your filename choice).  choose the name "a.json"
    - close Horizon EDA
    - delete the a.json file (because you are trying to create a test case and check that it works)
    - probably important: don't "update pool" at any point.

2. the steps in the original bug report:
    - open horizon, and the pool
    - select the parts section, create a part, change the name to something
    - click the "Save As" button, enter "a.json" as the filename, and save it
          - at this point the "Save As" button is still labelled as "Save As" **
    - close the parts window, observe the "Save Changes Before Closing?" dialog, pick an option.
    - Horizon EDA will now crash

When trying to reproduce this crash reliably, I know that usually using the "Save As" button will cause it to change the label to "Save" at the ** point.  Because that doesn't happen in the situation that will ultimately crash: it suggests that in some way the code 'knows' something is wrong.

It is also possible to cause a different sort of chaos by:
- Creating a part
- Press the Save As button and choose an existing filename (and save it)
- this time the "Save As" button becomes "Save" and you can close the part window without a crash
- But... both the new part and the (now overwritten) old part are in the pool manager

I originally cause the crash by not trying to mess things up (saving over an existing part, or trashing the pool).  My guess is that I originally caused this bug by creating a part, realising that I made a mistake and then 'tidying up' the filesystem (using my version software of choice) and then causing chaos.  In any case: this is clearly user error, but it would be nice if Horizon-EDA were resistant to crashing even if the user is being an absolute fool.

Maybe a solution would be for Horizon-EDA to refuse to save a 'new' json file if that filename is already indexed in the pool database?
(don't worry, I understand that user submitted solutions are often terrible ideas - so don't feel the need to spare my feelings at all).

carrotIndustries, 2021-12-26:
> Maybe a solution would be for Horizon-EDA to refuse to save a 'new' json file if that filename is already indexed in the pool database?

Seems reasonable, I've added a constraint to the database to enforce this. Needs some more testing before merging it into master.

> click the "Save As" button, enter "a.json" as the filename, and save it
> - at this point the "Save As" button is still labelled as "Save As" **

That's not supposed to happen: can you repeat this with the branch linked below and post the output? To get the output, you need to run `horizon-eda.exe` from the MSYS2 shell.

https://github.com/horizon-eda/horizon/runs/4637525374?check_suite_focus=true

matt-roberts, 2021-12-27:
I've recorded it crashing using OBS, here's a link:  [https://mattroberts.org/2021-12-27_13-17-52.mp4](https://mattroberts.org/2021-12-27_13-17-52.mp4)

The console log looks like this at the end of the recording:
> $ ./horizon-eda.exe
> 
> (horizon-eda.exe:18288): GLib-GIO-WARNING **: 13:17:55.785: win32 session dbus binary not found: C:\msys64\home\micro\horizon-run4637525374\\gdbus.exe
> hello from thread
> pool update took 5.776
> 380
> Segmentation fault

In the video: the background window on the left is the MSYS2 shell with the log messages that you wanted to see.  The background window on the right is an explorer view of the parts directory where the a.json _should_ have appeared.  The file was never written (I had the window visible so we could see exactly when the file appeared).

I think that there is a race condition going on.  It seems that when I take my time using the "Save As" button (and saving the json part file) there is a brief period before "Save As" button changes label to be "Save".  If I wait for that to happen, then I cannot cause the crash.  If I rush (as I did in this recording) to close the part window then I can cause the crash.

Because timing seems to matter, I have a computer game running in the background just to generally slow down my computer.  (The time window for rushing gets much wider if I take the time to alt-tab into the game and back again).

carrotIndustries, 2021-12-27:
It's strange that there is no output when you use "Save as" for the first time. I've tried it on a windows machine as well and the output shows up as to be expected.

> It seems that when I take my time using the "Save As" button (and saving the json part file) there is a brief period before "Save As" button changes label to be "Save"

That's really strange. I've added some more debug output in https://github.com/horizon-eda/horizon/actions/runs/1627754696 maybe something's wrong with the return code from the filechooser.

matt-roberts, 2021-12-27:
I have three videos for you (all have a frame rate of 30fps, which turns out to be potentially useful):

[https://mattroberts.org/2021-12-27_21-52-36.mp4](https://mattroberts.org/2021-12-27_21-52-36.mp4)
This run results in a segmentation fault.  I already had a part file called a.json (in error, but the crash still happens - so hopefully there is something to learn).  Anyway, the console log at the end looks like:

> $ ./horizon-eda.exe
>
> (horizon-eda.exe:856): GLib-GIO-WARNING **: 21:57:08.266: win32 session dbus binary not found: C:\msys64\home\micro\horizon-run4643825446\\gdbus.exe
> hello from thread
> pool update took 5.698
> 380
> 383
> 385
> 387
> Segmentation fault

[https://mattroberts.org/2021-12-27_21-55-14.mp4](https://mattroberts.org/2021-12-27_21-55-14.mp4)
This run worked.  You can see my surprise in the pause after it worked before I stop recording.  (After all, I am trying to do the same things each time to provoke the same bug each time.  But it isn't always the same).
If you look carefully, you can this sequence of things happens in about 0.1s:
- I click "Save As", leave the default filename "a.json" and click "Save" in the filename chooser
- The filename chooser closes, the "Save As" button remains "Save As"
- Then the logging happens in the console
- And the "Save As" button becomes the "Save" button
- A little while later the "a.json" file appears in the explorer window (although this delay is almost certainly the time it takes explorer to do its thing and redraw)

The console log looks like this at the end:

> $ ./horizon-eda.exe
>
> (horizon-eda.exe:20884): GLib-GIO-WARNING **: 21:57:08.266: win32 session dbus binary not found: C:\msys64\home\micro\horizon-run4643825446\\gdbus.exe
> hello from thread
> pool update took 5.841
> 380
> 383
> 385
> 387
> 389
> 391
> 393
> hello from thread
> 395
> 397
> pool update took 0.044

[https://mattroberts.org/2021-12-27_21-57-04.mp4](https://mattroberts.org/2021-12-27_21-57-04.mp4)
This run is the crash I think I am trying to cause.  (there is no preexisting a.json part file).  If you look at the timing of the console messages, you'll see that I can enter the filename from "Save As", close the part window and say "Don't Save Changes" before "380" appears in the console log.

The console log looks like this at the end:

> $ ./horizon-eda.exe
>
> (horizon-eda.exe:6092): GLib-GIO-WARNING **: 21:57:08.266: win32 session dbus binary not found: C:\msys64\home\micro\horizon-run4643825446\\gdbus.exe
> hello from thread
> pool update took 5.905
> 380
> 383
> 385
> Segmentation fault

carrotIndustries, 2021-12-27:
Did you run the old or new binaries? I think we should debug where it actually crashes:

In the Msys2 shell, run `gdb horizon-eda.exe`. In the gdb shell, enter `r` to run it. Once it has crashed, enter `bt` to show a backtrace. For best results, build it from source to get debug symbols: https://docs.horizon-eda.org/en/latest/build-win32.html

I'm also wondering why the output happens after you close the window and not immediately after saving.

matt-roberts, 2021-12-27:
> Did you run the old or new binaries?

The new binaries.  The directory in the MSYS2 window is my way to keep track.  ```~/horizon-run4643825446``` refers to magic number in this url: [https://github.com/horizon-eda/horizon/runs/4643825446?check_suite_focus=true](https://github.com/horizon-eda/horizon/runs/4643825446?check_suite_focus=true)

carrotIndustries, 2021-12-27:
> The new binaries.

These should print 379, 381, and no 380: https://github.com/horizon-eda/horizon/blob/b836342b79ed867991bcea766bc2cd1f0826d59f/src/pool-prj-mgr/pool-mgr/editors/editor_window.cpp#L379-L400

matt-roberts, 2021-12-28:
> These should print 379, 381, and no 380
Oh, I see what you mean.  Checks.... hangs head in shame: looking at CRC32 for the thing I actually ran, it was the old binary.
(looking in the downloads directory, it is obvious what I did.  I foolish thought that explorer would put the newest file first when it was in sections like "Today" and "Earlier This Month".  It turns out that it is ordered by filename in those sections, whcih turns out to the reverse time order than I expected. FML.)

Sorry for the confusion.  In any case I'm trying to follow the build instructions to generate debug version to generate the backtrace for you, hopefully that'll be more useful anyway!

matt-roberts, 2021-12-28:
I'm impressed with your build instructions.  They worked second time (and the first time I ran "MSYS2 MSYS" rather than "MSYS2 MinGW x64" you asked for, so that was on me).

Anyway, here is the (now traditional) video: [https://mattroberts.org/2021-12-28_10-38-08.mp4](https://mattroberts.org/2021-12-28_10-38-08.mp4)
I've taken a peek at the "about" screen at 0:21 so we are certain exactly what version I'm actually running.
The crash happens at about 1:30 and seems to be a quicker one.

The obviously useful stuff in the gdb window is:

> Thread 1 received signal SIGSEGV, Segmentation fault.
> horizon::EditorWindow::save (this=0x20fcaeed7d0) at src/pool-prj-mgr/pool-mgr/editors/editor_window.cpp:380
> 380                     iface->save();
> (gdb) bt
> #0  horizon::EditorWindow::save (this=0x20fcaeed7d0)
>     at src/pool-prj-mgr/pool-mgr/editors/editor_window.cpp:380

Hopefully I've attached a copy of the log here: [gdb.txt](https://github.com/horizon-eda/horizon/files/7784295/gdb.txt)

matt-roberts, 2021-12-28:
Before I closed the gdb window, I decided to (learn how to) have a peek at iface to see if it happened to be a null pointer or something:

>(gdb) print iface
> $1 = (horizon::PoolEditorInterface *) 0xfeeefeeefeeefeee

matt-roberts, 2021-12-28:
I was curious at that very suspicious value, so I did the tiniest amount of searching: Wikipedia suggests that 0xfeeefeee is the magic number used by Microsoft's debug HeapFree() to mark freed heap memory.
[https://en.wikipedia.org/wiki/Magic_number_(programming)#Debug_values](https://en.wikipedia.org/wiki/Magic_number_(programming)#Debug_values)

carrotIndustries, 2021-12-28:
> Before I closed the gdb window, I decided to (learn how to) have a peek at iface to see if it happened to be a null pointer or something:

Excellent.

That's really strange, I've added some more debug prints to the 637 debug branch. Just do the usual steps and post the output. I'm still a bit puzzled why I cant reproduce this behavior.

Thanks for your patience so far!

matt-roberts, 2021-12-28:
Okay.  I think I've compiled the correct version.  (both git status and the about dialog say ```ed7c08b8```).

Here is the video (should you care about the exact timings):  [https://mattroberts.org/2021-12-28_22-11-42.mp4](https://mattroberts.org/2021-12-28_22-11-42.mp4)
Here is the console log in full: [gdb.txt](https://github.com/horizon-eda/horizon/files/7786559/gdb.txt)
And the most likely bit you care about (that which appeared after I actually click "Create"):
```
252 EditorWindow 0x23df5752b18
350 save 0x23df5752b18
[Thread 19232.0x35e0 exited with code 0]
[Thread 19232.0x5750 exited with code 0]
[Thread 19232.0x5604 exited with code 0]
[Thread 19232.0x374c exited with code 0]
[Thread 19232.0xa1c exited with code 0]
[Thread 19232.0x5790 exited with code 0]
[Thread 19232.0x4b6c exited with code 0]
[New Thread 19232.0x471c]
[New Thread 19232.0x17e0]
[New Thread 19232.0x3d58]
[New Thread 19232.0x4f18]
383 save
[New Thread 19232.0xd74]
[New Thread 19232.0x3b30]
[New Thread 19232.0x35c0]
[Thread 19232.0x3d58 exited with code 0]
[New Thread 19232.0x52c8]
[Thread 19232.0x1c5c exited with code 0]
[Thread 19232.0x52c8 exited with code 0]
[New Thread 19232.0x2bd8]
[New Thread 19232.0x35c4]
[New Thread 19232.0x5410]
warning: onecore\com\combase\dcomrem\security.cxx(3071)\combase.dll!00007FFCE225904C: (caller: 00007FFCC1F1823E) ReturnHr(1) tid(5410) 80010117 Call context cannot be accessed after call completed.
[New Thread 19232.0x542c]
[New Thread 19232.0x1f14]
[New Thread 19232.0x4f78]
[New Thread 19232.0x180c]
[New Thread 19232.0x3b80]
[Thread 19232.0x3b80 exited with code 0]
warning: onecore\com\combase\dcomrem\resolver.cxx(2299)\combase.dll!00007FFCE223206D: (caller: 00007FFCE2234F4E) ReturnHr(2) tid(1974) 80070005 Access is denied.
warning: onecore\com\combase\dcomrem\resolver.cxx(2299)\combase.dll!00007FFCE223206D: (caller: 00007FFCE2234F4E) ReturnHr(3) tid(2bd8) 80070005 Access is denied.
warning: onecore\com\combase\dcomrem\resolver.cxx(2507)\combase.dll!00007FFCE2234F76: (caller: 00007FFCE2231A05) ReturnHr(5) tid(1974) 80070005 Access is denied.
warning: onecore\com\combase\dcomrem\resolver.cxx(2507)\combase.dll!00007FFCE2234F76: (caller: 00007FFCE2231A05) ReturnHr(4) tid(2bd8) 80070005 Access is denied.
[New Thread 19232.0x2a08]
[Thread 19232.0x2a08 exited with code 0]
[New Thread 19232.0x21bc]
271 operator() 0x23df5752b18
~EditorWindow
[Thread 19232.0x3b30 exited with code 0]
[Thread 19232.0x21bc exited with code 0]
385 save -3
387
390 save
392 save

Thread 1 received signal SIGSEGV, Segmentation fault.
0x00007ffcaea1397f in ?? () from D:\msys64\mingw64\bin\libstdc++-6.dll
(gdb) bt
#0  0x00007ffcaea1397f in ?? () from D:\msys64\mingw64\bin\libstdc++-6.dll
#1  0x00007ff638a4ca78 in std::__cxx11::list<sigc::slot_base, std::allocator<sigc::slot_base> >::emplace<sigc::slot_base> (__position=...,
    this=0x7ffc8239c468)
    at D:/msys64/mingw64/include/c++/11.2.0/bits/stl_list.h:200
#2  std::__cxx11::list<sigc::slot_base, std::allocator<sigc::slot_base> >::insert (__x=..., __position=..., this=0x7ffc8239c468)
    at D:/msys64/mingw64/include/c++/11.2.0/bits/stl_list.h:1309
#3  sigc::internal::temp_slot_list::temp_slot_list (
    slots=std::__cxx11::list<error reading variable: Cannot access memory at address 0x200008123>, this=<synthetic pointer>)
    at D:/msys64/mingw64/include/sigc++-2.0/sigc++/signal_base.h:223
#4  sigc::internal::signal_emit1<void, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, sigc::nil>::emit (_A_a1=...,
    impl=0x7ffc8239c460)
    at D:/msys64/mingw64/include/sigc++-2.0/sigc++/signal.h:1035
#5  sigc::signal1<void, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, sigc::nil>::emit (_A_a1=..., this=0x23df2271470)
    at D:/msys64/mingw64/include/sigc++-2.0/sigc++/signal.h:2951
#6  horizon::EditorWindow::save (this=0x23df22713b0)
    at src/pool-prj-mgr/pool-mgr/editors/editor_window.cpp:393
```

carrotIndustries, 2021-12-29:
Hmm, now at least I have a vague understanding of what might be going on:

- `gtk_native_dialog_run` enters a recursive main loop for the modal file chooser dialog
- you select the file and save it
- now things go wrong: the recursive main loop is still running and receives the "window close" event that ultimately deletes and frees the window
- the recursive main loop exits and the code continues running
- it tries to use members of the deleted window and crashes

Some more things to try:

 - Wait up to a minute or so after saving and see if the button eventually shows "Save". Only then we're back to the correct main loop.
 - It seems you've got some network drives mounted, one of which isn't accessible. Perhaps this causes the filechooser loop to hang

After way too much fiddling I was finally able to reproduce this bug and confirm the above suspicions. I had to map a network drive, then disconnect the network, trigger a refresh of the inaccessible network drive in windows explorer and quickly open the save dialog. This got me into the same situation you are in. From the backtrace I could tell that it indeed deleted the window from within the filechooser event loop.

I've added a workaround that disables the whole part editor window until the file chooser loop is done. This should prevent closing the window from within the filechooser loop.

Finally, this needs to be fixed properly in Gtk, I'll try to come up with a minimum working example so I can open a bug report with them.

carrotIndustries, 2021-12-29:
> Finally, this needs to be fixed properly in Gtk, I'll try to come up with a minimum working example so I can open a bug report with them.

Let's see what happens: https://gitlab.gnome.org/GNOME/gtk/-/issues/4582

matt-roberts, 2021-12-29:
I tried pausing after saving.  That seems to not crash.  Video is here:

[https://mattroberts.org/2021-12-29_18-32-22.mp4](https://mattroberts.org/2021-12-29_18-32-22.mp4)

I noticed that after pressing "Save" in the file chooser:
- I didn't interact with the part window for a bit (a few seconds)
- Then the "Save As" button became a "Save" button
- And at the moment that happened, a bunch of console messages appeared

I think that this confirms that we've worked out what's going on.

(Next step, I'll turn on the machine that the network drive is on, allowing it to connect, and see if the need to pause goes away)

matt-roberts, 2021-12-29:
I turned the machine at the other end on, and then used explorer to browse the files on it.  So I can say that the network drive is fully working.

And, as expected, the Save As button turns into the Save button without anywhere near the same delay (it went from multiple seconds to maybe 1/4s).

matt-roberts, 2021-12-29:
I've just compiled and tested ```835afff3d0d6e9f19e411c73bc180e9b3ffef91d```.  And it behaves exactly as you'd expect: if I try to close the part window after saving (but before the "Save As" has become "Save") my click is ignored.  When the "Save As" button becomes the "Save" button, I can close the part window without crashing, and the part json file is actually saved.

So the workaround works.

I'm guessing that it'd be possible to have the same problem everywhere that you ask gtk to come up with a filename... which might mean littering your code with this workaround in many more places?

carrotIndustries, 2021-12-30:
> I'm guessing that it'd be possible to have the same problem everywhere that you ask gtk to come up with a filename... which might mean littering your code with this workaround in many more places?

I've put this in it's own function and used search/replace to substitute all calls to `gtk_native_dialog_run` with this wrapper, so it doesn't harm the other code too much. Once it's fixed in Gtk, this commit should be fairly easy to revert.

Let me know if it works, so we can close this issue. Thanks again for your patience in helping to debug such an elusive problem.

closed 2022-01-03 by carrotIndustries

matt-roberts, 2022-01-04:
I confirm that I cannot cause the crash anymore.  (Sorry it took so long to get back to you, but I was using your software productively!)

We probably still want to talk about this:

>> Maybe a solution would be for Horizon-EDA to refuse to save a 'new' json file if that filename is already indexed in the pool database?
>
> Seems reasonable, I've added a constraint to the database to enforce this. Needs some more testing before merging it into master.

I'll collect my thoughts, and open a new issue.

commit d44f402e (2022-01-03): win32: use patched gtk

includes a fix for https://gitlab.gnome.org/GNOME/gtk/-/merge_requests/4326

closes #637