horizon-eda/horizon #640: Feature Request: Add ability to highlight all places with the same part in the 3d view

frmdstryr, 2021-12-30:
I'm not sure how many others assemble boards themselves but I was doing a few boards today and thought it would be helpful if the 3d view had a way to highlight all places where a given part is used.

Eg If I click one 10k resistor, have a keyboard shortcut or something to show where all the other 10k resistors are.

Somewhat related, it might also be helpful to have an option to show the BOM or some sort of "assembly assistant" in the 3d view.

Any thoughts?

carrotIndustries, 2021-12-30:
I think you're looking for the parts window, it's available from the spacebar menu: https://blog.horizon-eda.org/progress/2020/08/29/progress-2020-07-08.html#parts-window

closed 2021-12-30 by carrotIndustries

frmdstryr, 2021-12-30:
Ah, that's perfect!  

Is there a button in the 3d view to open it? I found it in the spacebar menu but it wasn't exactly obvious without knowing it exists..

carrotIndustries, 2021-12-31:
> Is there a button in the 3d view to open it?

Nope, it's only in the menu as I couldn't think of any other place to put it.

frmdstryr, 2021-12-31:
I think adding an entry for it in the dropdown menu where the "Fabrication output" is would be helpful.

carrotIndustries, 2021-12-31:
See 4d436e46ae3863fa8cda6b15e99eef32f5d418b5. I took this as an opportunity to rearrange the hamburger menus so that the order makes some sense.

frmdstryr, 2021-12-31:
Looks good to me, thank you!!