horizon-eda/horizon #643: Consistency of the pool database with files on disk

matt-roberts, 2022-01-04:
_Found when I was trying to explore what caused issue [637](https://github.com/horizon-eda/horizon/issues/637), but this isn't that at all._

If you save a new part with the same filename as an existing part, then the pool manager gets out of sync with the files on disk.

Steps to repeat:
1. In the pool manager, select "Parts", click "Create"
2. Get to the point of saving the new part
3. Select the same filename as an existing part in the pool, and save

What happens (rev ```835afff3``` on Windows):
during step 3, there is a dialog box (from gtk or windows) saying "File already exists, Do you want to replace it?", obviously I clicked "Yes"
then Horizon-EDA says*** "Pool update error", "UNIQUE constraint failed: parts.filename, parts.pool_uuid"
at this point:
- the new part exists on disk, but not the old part (this is exactly what I asked for)
- the pool manager has the old part present, but not the new part
- if you try to select the old part in the pool manager, that causes an "Exception in signal handler" (Part _some UUID_ not found)
- you can fix this this by clicking "Update Pool"

*** this "pool update error" is new since v2.1, and is certainly an improvement.  It would certainly help any user to realize that something is up.

While I understand that user suggested fixes are often terrible... I think there are two obvious (to me) things that would 'work'.  (by 'work', I mean that after the dust has settled, the files on disk and the pool manager are consistent).

> when Horizon-EDA actually saves the part for the first time; check if the part filename is in the pool database and either:
> - remove that record first.
> - or don't save the part file and throw a dialog box to the user explaining what's going on.

The first of these would hopefully result in the same files on disk as now, but a pool database that matches the files
The second of these would stop the chaos from happening in the first place.  Although it'll be slightly jarring to be asked "do you want to replace" and then be told "sorry, but no".  But that is a small price to pay to avoid chaos.

(Of course, I'm actually half expecting to be told of a much better idea that didn't occur to me at all.  I guess I'm saying "don't spare my feelings").

carrotIndustries, 2022-01-12:
The linked commit fixes this by falling back to a complete pool update if that happens.

closed 2022-01-12 by carrotIndustries

matt-roberts, 2022-01-13:
Confirmed working at this end.

commit bae8ce00 (2022-01-12): pool update: perform complete pool update when files got overwritten

fixes #643