horizon-eda/horizon #647: Assertion failed: (std::abs(da) <= M_PI / 2), function pdf_arc_segment, file src/export_pdf/canvas_pdf.cpp, line 209

unrelentingtech, 2022-01-07:
Hit this assertion when exporting the Outline layer on my current project.

With

```diff
--- i/src/export_pdf/canvas_pdf.cpp
+++ w/src/export_pdf/canvas_pdf.cpp
@@ -206,7 +206,10 @@ static Coordd pdf_arc_segment(PoDoFo::PdfPainter &painter, const Coordd c, const
 {
     const auto da = a0 - a1;
     assert(da != 0);
-    assert(std::abs(da) <= M_PI / 2);
+    if (!(std::abs(da) <= M_PI / 2)) {
+        std::cout << "a0 " << a0 << ", a1 " << a1 << ", r " << r << ", c " << c.x << ";" << c.y << std::endl;
+        return c;
+    }
 
     // Shift to bisect at x axis
     const auto theta = (a0 + a1) / 2;
```

the output is

```
a0 -1.03308, a1 -2.60388, r 11.3386, c 70.2113;-222.031
```

and the resulting file is [this.pdf](https://github.com/horizon-eda/horizon/files/7830011/b.pdf)

so I guess it's the bottom of the right PCB — that's where the PDF renderer drew a straight line instead of the arc that's supposed to look like this:

[image]

carrotIndustries, 2022-01-07:
can you share a minimum working example to reproduce this? pasting the polygon in question into a new project should to the trick

@frmdstryr, can you have a look at this

frmdstryr, 2022-01-07:
Seems like it's something with rounding error?   
```python
a0 = -1.03308
a1 = -2.60388
abs(a0-a1)
Out[19]: 1.5708000000000002
pi/2
Out[20]: 1.5707963267948966
```

Perhaps a tolerance should be given in the check, eg `std::abs(da) <= M_PI/2 + 1e-6` or something?

carrotIndustries, 2022-01-07:
> Perhaps a tolerance should be given in the check, eg std::abs(da) <= M_PI/2 + 1e-6 or something?

Dunno, does the bezier calculation go haywire if it's slightly more than PI/2? Or maybe make `da` in

https://github.com/horizon-eda/horizon/blob/31fbd6f1a4aac02568444c9d3340d0f8cf771f79/src/export_pdf/canvas_pdf.cpp#L247

a tad smaller than pi/2?

frmdstryr, 2022-01-07:
I don't think so, there will just be some error in the circle approximation if it's slightly over. See https://en.wikipedia.org/wiki/B%C3%A9zier_curve#Properties.

frmdstryr, 2022-01-07:
Or maybe the tolerance should be at:

https://github.com/horizon-eda/horizon/blob/09e0e854d730a0f8dff635681139fd3f981bca9c/src/export_pdf/canvas_pdf.cpp#L255

unrelentingtech, 2022-01-07:
Here's a project (without a cache pool, inserting the pool from a fresh project should work I guess) [repro-issue647.tar.gz](https://github.com/horizon-eda/horizon/files/7830469/repro-issue647.tar.gz)

> Perhaps a tolerance should be given in the check, eg std::abs(da) <= M_PI/2 + 1e-6 or something?

Usually floating point comparisons should include this… And yeah this particular modification does help :)

carrotIndustries, 2022-01-07:
closed in #648

closed 2022-01-07 by carrotIndustries