horizon-eda/horizon #653: Thoughts on the impact of step models in the pool

atoav, 2022-02-21:
The 3D-Step files make up the biggest majority of the current official pool. To be a little bit more precise roughly three quarters:

[image]

This could become a problem in the long term, just because of the size. 

In terms of functionality these 3/4 of the size are "just" there for the 3D-previews – something that is great to have, but not necessarily the most crucial core functionality used by most EDA users. Even if a user _loves_ the 3D-functionality, a typical user might maybe see only a very small percentage of 3D models.

Having the 3D models being part of the pool makes sense from a version management perspective, but my gut feeling is that in the long run some other way where models are only downloaded when they are needed – or lazily in the background when they are added to the schematic. For this they have to be kept seperate from the pool however.

carrotIndustries, 2022-02-21:
Thanks for bringing this up, I also believe that the current approach isn't sustainable on the long run unless the size of the pool isn't an issue for people. Compressing will only buy us some time and we'll have the same discussion in some years.

> In terms of functionality these 3/4 of the size are "just" there for the 3D-preview

And for the STEP export.

The KiCad 3d models library is currently >5GB in size (though some of that is due to the WRL files), so i think downloading all 3D models is something we should avoid if possible.

That's why I think the best approach is do lazily download the 3D models as needed. So here's a brain dump of how that could look like:

Next to each `.step` file, there's a `.step.ref` JSON file that stores how the 3D model can be obtained and a hash of the 3D model to ensure consistency, though I'm not quite sure if we really need the hash part. So far I've thought of these methods for getting the 3D models:

# `pool`
The `pool.json` file will contain a base url similar to `https://github.com/horizon-eda/horizon-pool/blob/` and the `.ref` file contains the path to the 3D model. It's still to be decided if the URL should point to a particular branch or commit:

pro branch:
 - We can just put the default branch name in the base URL and use the path the `.ref` file is at to produce the complete URL.
 - No messing with commit hashes

pro commit:
- Hashes in the ref file won't break if someone is on an older commit of the pool repo and the 3D model in the 3d repo has changed.
- Makes it possible to accurately reconstruct a point in time in the pool repo.

Another approach to this method could be having an action in the 3d models repo that keeps an index of hash to filename so the 3D models are addressed by hash/content rather than filename. This would also eliminate the issues associated around renaming/moving 3d models. Over just using the hash as a filename, this has the advantage that it's nicer to humans.

# 'http`
The `.ref` file contains a complete http URL to download the 3D model. Can also be used to download 3D models with a license that's incompatible with the pool

# `manual`

Some 3D models can't be downloaded without manual interaction, such as registration forms and other nonsense that vendors come up with.

---------

I've though about adding the last two methods independently of moving 3D models out of the pool to make it easier for people to use vendor-provided 3D models.  

The pool manager will then grow a 3D models tab to provide an overview of the download status.

The github PR integration in the pool manager could then take care of opening a PR in the 3D models repo as well and a bot could take care of automatically merging the PR in the 3D models repo.

I don't quite have an idea yet of what to make of the existing pool repo. Either we keep it moving it along with the 3D models it accumulated over time or we do something breaking such as making a new repo, creating a new orphan branch without the 3D models...

No matter what we do, it'll and up adding more complexity, but I don't really see a way around it unless we're fine with the pool repo eventually being gigabytes in size. @fruchti @RX14  @jue89 any thoughts on this?

tkornack, 2022-02-21:
Would this be a good fit for git-lfs?

https://git-lfs.github.com

atoav, 2022-02-22:
I think I like the http approach most, because it is less dependend on the intricates of git. If a regular http GET is used however, I'd definitly advice for storing the hash in the pool as well so it can be compared against the downloaded file.

### The downside of relying on HTTP is of course that there has to be some handling for:

- Local caching (Storage is cheap, I'd say, just cache it forever/till it is modified/moved/deleted)
- Gracefully handle all kind of HTTP Errors, Timeouts, etc.
- Hash comparison, handling of a failed hash comparison
- Unable deal with situations where someone has an outdated pool with old hashes pointing to updated resources. Maybe a pool update can be suggested then?

### On the pro side it would:
- allow for more sources
- fix some licensing headaches with the pool
- be conceptually understandable (one URL = one 3D-file)

### Additionally:
- Maybe we could make the URL parameter of the `.ref` a _list_ of URLs to allow multiple sources for reliabilty?

## as for the existing pool

I think the wise decision would be to remove step files completely from the pool and only keep the ref files there. The precise way in which this has to be done has to be figured out of course. If we use http for the step files, it doesn't matter at all where we are storing them, as long as the bandwidth is there, it is reliable and doesn't demand a ton of maintenance. It could as well be another git repo.

## Open questions
- How would contribution work? Put the 3D-part in the 3D-part repo and make a PR?
- How would this affect the in-programme-git-pane? Maybe step files added to the main pool can then be automatically hashed, `.ref` files created, the step files moved to their own repo, pushed as well?

All in all I have the feeling this might not be entirely trivial, but it is certainly not as hairy as I though it could be.

atoav, 2022-02-22:
> Would this be a good fit for git-lfs?
> 
> https://git-lfs.github.com

This could be worth a look as well huh.

jue89, 2022-02-22:
TBH - I would leave it like it is. Disk space is sufficient nowadays. Internet is fast in most parts of the world. (Maybe some people from Brandenburg, Germany would need certain time. But this pain will drive future bandwidth extensions due to the demand ...) And people are already downloading games with 50GB+ over the Internet. 5GB isn't that heavy like it was 10 years ago. I'd say usual disk sizes und bandwidth will increase in future even more. They'll grow with the pool size, I'd say ...

I'm concerned that introducing a distinct place for step files just creates a lot of complexity:

* Mechanics have to be integrated for lazy-loading und caching.
* More infrastructure has to be maintained: Beside a git server, we would need a HTTP server (or a git server with lfs enabled ...). That will hinder people that may want to have their own pool without any connection to the original pool.
* Working offline isn't that easy anymore. Horizon will need a permanent connection to the Internet while designing boards.
* People may will struggle using old pools in the future, because those step file serving machines aren't online anymore.

But maybe I'm underestimating the problem. Has anyone faced actual problems with big pools, yet? Once the pool is downloaded, pulls aren't a time-consuming operations anymore.

atoav, 2022-02-22:
> That will hinder people that may want to have their own pool without any connection to the original pool.

There is no necessity for this to be an online-only thing – the URL could be just an _additional_ way of serving step files. 

The reasoning from my side is mostly about the repo and the management thereof. Couldn't a way to reference/load parts from outside the pool make sense from that perspective? Whether the parts in that outside location are downloaded on demand, or in full lazily in the background as soon as you download the official pool is worth a discussion and could (ultimately) even be a setting in the preferences. The point is that forking, cloning etc. the pool isn't cloning all that 3D-data as well – also because it is not really editable *source data* from the horizon perspective and because it might potentially have a chilling effect on contributions with a lot of 3D data.

> People may will struggle using old pools in the future, because those step file serving machines aren't online anymore.

Technically we have the same issue right now with the URLs of datasheets. The way this is dealt with is to disallow certain unreliable sources by convention. But let's not make this an bigger issue than it is. If a 3D-model doesn't load that might be unfortunate, but ultimately it shouldn't fundamentally impact your ability to make use of an EDA software. 

There are many packages today that do not have a 3D-model attached to them, although you might find one on the manufacturer side. In those cases external sources would make the difference between having the chance to get a 3D model and having none at all. If it is locally cached your old projects should still be able to load the old 3rd part models even when the manufacture went bust or their webserver is on fire.

> Mechanics have to be integrated for lazy-loading und caching.

True. Maybe I am biased, because I deal with such things daily (although not in C++) and I don't see them as such a huge deal to implement these days. The caching must not be overly sophisticated, as you rightly said: storage is cheap.

> Once the pool is downloaded, pulls aren't a time-consuming operations anymore.

To be quite honest here, this is also about the *feeling* for me. I like architectures where I don't have to worry too much about the impact the 3D-model of my tiny insignificant part might have on the whole structure in the long run – this is especially true when I create them via script like in [ this PR where I add 47 step files with a whopping 200+ MB](https://github.com/horizon-eda/horizon-pool/pull/282/files). Connectors that have a different shape for each number of pins require a 3D-model per package – having 3D models for them also makes sense, because you need to think about the space around them a lot.

If we were to keep things as they are any reasonable maintainer would rightfully conclude that 200+ MB of 3D data is a bit too much for something as insignificant as box headers. Which would then mean I won't add those 3D-files, which would then mean nobody can use them, etc.

This means having the goal to find a reasonable, maintainable, clever solution that helps avoiding that problem is the goal of that issue. I am convinced there are ways of doing this that would be a real improvement on all fronts without removing existing and working ways of doing things.

jue89, 2022-02-22:
These are all good reasons, for sure.

But I have to force myself to opt for the most practical solution than the one that is really clever. Solutions that computer scientists call *clever* are *complex* solutions for non-computer scientists in most cases. Never underestimate the *stupid simple* solution.

But of course, I wouldn't stop using Horizon just because step files aren't shipped with the pool repo. (I'm more a computer scientist loving *clever solutions* than the electrical engineer who just needs a tool for designing PCBs. And if I would, I still would throw PCBs together with Kicad and just don't care about funky on-disk formats and inconsistent library architectures.)

carrotIndustries, 2022-02-22:
> Would this be a good fit for git-lfs?

I don't think so: LFS only helps if big (binary) files often change, so a clone doesn't have to download all past versions of these files. In our case, 3D models, shouldn't change that often once they're in the pool.

Some clarifications on how the downloading could work: Once 3D models is downloaded, it's put in the same place as it would have been if it was in the pool. The only thing that will need to change is that retrieving a 3D model from the pool can return a "please try again later" sentinel value and the pool has a signal that is emitted once a 3D model has become available. That way, all of the mess that comes with interacting with the outside world is confined to a single place where it can be dealt with appropriately. Also, downloaded 3D models will stay there and there could as well be an option do download everything in bulk, so offline use will still be possible.

The primary point for managing the `.ref` files could be the package editor since that's the place where the 3D model is encountered first. 

> More infrastructure has to be maintained: Beside a git server, we would need a HTTP server

Most if not all hosted git implementations provide means of downloading files from repos without any extra maintenance effort. 

> People may will struggle using old pools in the future, because those step file serving machines aren't online anymore.

For 3D models that are now part of the pool, that would just be another repo. By referencing the file by hash and not deleting files in the 3D model pool, I think we can avoid this in all reasonable circumstances.

For vendor-provided 3D models in the excluded `local` directory, the `.ref` files would help others to get the model without manually browsing the vendor's website.

> Technically we have the same issue right now with the URLs of datasheets

I've once thought of a scrubber bot running every couple of days that scans the pool for broken links, maintains an overview of the status of all items and opens an issue if a datasheet link gets broken...

fruchti, 2022-02-22:
A lot of good points. I personally agree that there should be externalised 3D models of some kind but I am undecided which would be the better long-term solution. Adding referencing proprietary 3D models (for vendors allowing automatic download of their hosted files) to the mix does indeed sound like a very good ‘two birds one stone’ situation, but I am not sure if the complexity is worth it.

So I don’t have a clear favourite solution, but maybe two small points to consider:

IIRC GIT_LFS_SKIP_SMUDGE lets you skip the download of git-lfs tracked files when cloning a repository but I don’t know if more fine-grained control is possible with git-lfs. If it isn’t, this could be a direct, more straight-forward solution than a model with two repositories.

Automatic or semi-automatic HTTP downloads are really interesting for non-free models. I agree with @atoav in that a hash is a must, because a vendor could update the model with a new licence or a potential STEP parser bug could be exploited via a malicious model update otherwise. With non-free models in mind, a link to the licence/terms and conditions has to be included, especially since Horizon automatically copies models to a project’s cache.

atoav, 2022-02-22:
>  a potential STEP parser bug could be exploited via a malicious model update otherwise

this was one of my thoughts as well, I wonder however if the same would not also be possible by an entirely legit commit? My trust in step parsers is... limited.

> I've once thought of a scrubber bot running every couple of days that scans the pool for broken links, maintains an overview of the status of all items and opens an issue if a datasheet link gets broken...

Sounds like a decent enough idea. I don't know a lot about github bots, but if I can write a function in python that takes a pool and spits out a list of broken links and some stats I'd happily help with the effort.

RX14, 2022-02-23:
I think the solution @carrotIndustries already described is a very good starting point. As long as caching is persistent and not in tmpfs, it allows a great deal of offline working.

tkornack, 2022-02-23:
> > Would this be a good fit for git-lfs?
> 
> I don't think so: LFS only helps if big (binary) files often change, so a clone doesn't have to download all past versions of these files. In our case, 3D models, shouldn't change that often once they're in the pool.

Right, sorry, what about a partial clone? 

https://git-scm.com/docs/partial-clone

It seems appealing to keep all the relevant data together in a pool. The clone could also be shallow if it isn't already.