horizon-eda/horizon #654: Kicad package import fails on RPi_Pico_SMD_TH

jsgf, 2022-02-21:
The import fails with a bunch of unsupported pad stacks, but also a Critical error "SEXPR is not a double type".
From a process of elimination, this appears to be from lines will `(drill (offset ...))`:
```
  (pad 1 smd rect (at -8.89 -24.13) (size 3.5 1.7) (drill (offset -0.9 0)) (layers F.Cu F.Mask))
```

Full set of errors:
```
Seq | Level   | Domain                  | Message; Detail
0   | Warning | Import                  | Pad '3' is unsupported; type=thru_hole shape=rect
1   | Warning | Import                  | Pad '8' is unsupported; type=thru_hole shape=rect
2   | Warning | Import                  | Pad '13' is unsupported; type=thru_hole shape=rect
3   | Warning | Import                  | Pad '18' is unsupported; type=thru_hole shape=rect
4   | Warning | Import                  | Pad '23' is unsupported; type=thru_hole shape=rect
5   | Warning | Import                  | Pad '28' is unsupported; type=thru_hole shape=rect
6   | Warning | Import                  | Pad '33' is unsupported; type=thru_hole shape=rect
7   | Warning | Import                  | Pad '38' is unsupported; type=thru_hole shape=rect
0   | Critical| Import                  | exception: SEXPR is not a double type!; 
```
File in question:

[RPi_Pico_SMD_TH.kicad_mod.txt](https://github.com/horizon-eda/horizon/files/8112455/RPi_Pico_SMD_TH.kicad_mod.txt)

carrotIndustries, 2022-02-23:
This is because offset drills aren't supported yet and the importer assumes that the drill is specified as a single double rather than another expression.

jsgf, 2022-02-28:
Yeah that's what I thought. Should I leave this open as a TODO/feature request?

carrotIndustries, 2022-03-02:
> Should I leave this open as a TODO/feature request?

Yes, even though supporting it would require the kicad package importer to create its own padstacks.