horizon-eda/horizon #655: Round off convex corner

tkornack, 2022-04-07:
It seems that the wonderful "Round off vertex" tool doesn't handle convex corners. This is what happens when I try it:

[image]

Thanks for making horizon!

carrotIndustries, 2022-04-07:
>. This is what happens when I try it:

Yeah, back then I couldn't quite figure out how to pick the arc orientation. Press <kbd>e</kbd> to flip the arc in the round off vertex tool.

twarge, 2022-04-07:
oh, thanks!

closed 2022-04-07 by tkornack

carrotIndustries, 2022-04-07:
> Yeah, back then I couldn't quite figure out how to pick the arc orientation

I had another thought about this and believe that the new implementation correctly handles all cases.

commit e21595d0 (2022-04-07): tool round off vertex: fix arc orientation for non-convex corners

fixes #655