horizon-eda/horizon #656: Launching a editor window problem

joshuawins, 2022-04-08:
Launching a editor window shows exception dialog on windows 10.

Glib::Error: Failed to execute child process(Bad file descriptor).

When I launched using command line, it started editor window without problem in mingw64 bash prompt.
horizon-imp -d \\lib\\horizon-pool\\decals\\esd-silk.json (works fine!)

carrotIndustries, 2022-04-08:
Since when does this happen? I suggest you to use [procmon](https://docs.microsoft.com/en-us/sysinternals/downloads/procmon) to see what horizon-eda.exe is doing and where it goes wrong.

joshuawins, 2022-04-11:
**I have updated mingw64 libraries to install libarchive.**
$ pacman -Qs glibmm
local/mingw-w64-x86_64-glibmm 2.66.2-1

$ pacman -Qs libarchive
local/libarchive 3.5.3-1 (libraries)
local/mingw-w64-x86_64-libarchive 3.6.0-1

**after I launch a decal editing window(other editing windows also have the same problem) git can see the following messages on the terminal and got exception dialog.**
$ /c/tools/Horizon/horizon-eda.exe
hello from thread
spawn C:\tools\Horizon\horizon-imp -b M:\PCB\Physics\KSGV2\board.json M:\PCB\Physics\KSGV2\blocks.json M:\PCB\Physics\KSGV2\pictures

**I have captured processmon log until I got a exception dialog, but I cannot guess the problem. Anyway I have attached it.*
[Logfile.CSV](https://github.com/horizon-eda/horizon/files/8460350/Logfile.CSV)
*

carrotIndustries, 2022-04-11:
Can you also include `gspawn-win64-helper.exe` in the capture? You might also try setting `"capture_output"` to `false` to the `prefs.json` in `appdata/local(or roaming)/horizon`

joshuawins, 2022-04-12:
Installing version 2.2 and updating compiled files solved problem.
I was keep updating only the executables since the 1.x version.
Thank you for your kind support.
You did wonderful job!