horizon-eda/horizon #658: Feature request: optionally export copper with board STEP model

tkornack, 2022-05-06:
It would be great if the exported model could optionally match the 3D preview.

carrotIndustries, 2022-05-14:
As in including copper features? I once imported such a board into FreeCAD and it took forever. What's your use case for including copper features in the STEP model?

tkornack, 2022-05-14:
Yes, the copper features. I like to use copper-only features that need to be lined up with other geometry. I realize that FreeCAD isn't great for high complexity designs, but usually the boards in my application are very small and simple.

atoav, 2022-06-28:
@tkornack If it is "just" for visualization purposes, I exported the copper as a PDF, created textures for Blender in it used the image as a normal map). If done correctly this can yield you photorealistic results for the finished PCB.

carrotIndustries, 2022-06-28:
This is a feature that I'm not opposed to, but not motivated enough to implement myself right now. All of the building blocks are there:

 - Use `CanvasPatch` to flatten the board into a set of polygons, merge all copper patches on each layer into one `ClipperLib::PolyTree` object
 - Use https://github.com/horizon-eda/horizon/blob/545ca3e5b05aa625b9426484135732f465b063f6/src/export_step/export_step.cpp#L64 to turn each polygon/hole into a face.
 - Use `BRepAlgoAPI_Cut` (see above) to create faces with holes (there might be an easier way, but it's opencasade...)
 - Use `BRepPrimAPI_MakePrism` to turn the flat face into a prism of the right thickness
 - Use https://github.com/horizon-eda/horizon/blob/df45e445e677cfa542b44636d178c73e62699cee/src/export_step/export_step.cpp#L456 to add the resulting prisms to the assembly, give them meaningful names an colors

oxplot, 2022-12-02:
Another use is as product renders when imported in other 3D modelling packages.

atoav, 2022-12-02:
> Another use is as product renders when imported in other 3D modelling packages.

For product renders I had good results using image textures generated from the PDFs one can export. Blurring the images a little bit gives the impression of the traces being beneath a layer of solder stop paint if you use them as a bump-map.

Getting the trace geometry would already be possible as well via that route (PDF-export > convert to SVG > import SVG in Blender)

aa956, 2022-12-13:
More use cases for the accurate copper representation in 3D:

- Thermal simulation (so contrary to render it's useful to have actual copper thickness, all of the layers exported, accurate via models)
- CNC routing the PCB (here probably limited by 2-layer boards)

E.g. for kicad this is the project I'm using when actual copper in the model is required is https://github.com/realthunder/fcad_pcb

bsilver8192, 2022-12-31:
Silkscreen and soldermask can also both be helpful for designing and aligning mechanical components. For example, I've got some bus bars that bolt to the PCB. I'm going to keep the geometry simple enough to translate it over manually, but being able to view annotations from the PCB design to make sure I get everything in the right place so it lines up with the copper plane and soldermask polygon would be convenient.