horizon-eda/horizon #659: Feature request: resizable columns in various pool manager tables

twarge, 2022-05-15:
Some of the columns in the tables in the pool manager are not resizable for me. It would be handy to allow a wider column for the pin names in the unit editor for long FPGA pin names. 

[image]

carrotIndustries, 2022-05-15:
If the pin names are really long, they probably contain multiple functions that should be added as alternate names.

Making the columns resizable isn't quite straightforward unfortunately as the pin list is made from boxes and size groups rather than a treeview.

twarge, 2022-05-15:
OK, thank you, I see. I will follow your 10M25 example.

closed 2022-05-15 by twarge