horizon-eda/horizon #660: Feature request: Blind and burried vias

jue89, 2022-05-19:
Some ICs are just available in BGA packages with high density pitch. To connect those ICs blind vias are required which allow for smaller diameters than standard vias.

Some ideas how to design the UI/UX:

1. Blind and buried vias aren't allowed between arbitrary layers. I suggest, that vias can be defined in the Stackup configuration window. Every via type get its range of layers. For example for a 6 layer board, we could have 4 different via types: standard via layer 1-6, blind via 1-2, buried via 2-5, blind via 5-6. (Or shall vias get their own configuration window?)
2. Currently, when traces are drawn, vias can be placed with a hotkey (`v` by default). I'd suggest that the via types are toggled by pressing `v` again. For example, if we start drawing a trace on layer 1, the first selected via is the standard via. If we press `v` again it toggles to the blind via (1-2). All other via types aren't accessible on layer 1.
3. The rules for vias in the rules window can be set by via type.

Maybe this inspires you. Feel free to ask for feedback or if something shall be clarified.

SonoQuium, 2022-05-24:
Nice suggestion.

tkornack, 2024-05-10:
Close this now that the feature exists?

closed 2024-05-10 by carrotIndustries