horizon-eda/horizon #661: Feature request: Please Add Mouse Wheel functionality

SonoQuium, 2022-05-23:
As it is now the mouse wheel has only one functionality: Zoom.

Can mouse wheel functionality be added?
Mouse Wheel & Ctrl: Pan Horizontally
Mouse Wheel & Alt: Pan Vertically

Thanks.

SonoQuium, 2022-05-24:
There are 2 reasons that come to mind:
1. Because the 4 implemented methods you mention are restricted by the area that is shown on the canvas. This becomes apparent while zoomed-in pretty far. and with any of the implemented pan method I would have to take multiple steps like zooming out first and then pan. 
2. Because new potential users are used to it and love it...

carrotIndustries, 2022-05-24:
>  Because the 4 implemented methods you mention are restricted by the area that is shown on the canvas.

Both dragging with the left or middle mouse button aren't restricted to the canvas since the cursor wraps around at the edge of the canvas, enabling infinite panning. Trackpoint and touchpad scrolling also don't have any limitations.

All in all, I'm not convinced that this is a feature worth adding and supporting. If more users show up with that request, I might change my mind.

closed 2022-05-24 by carrotIndustries