horizon-eda/horizon #667: Rethink Procedure of (Gtk) Windows Closing

SonoQuium, 2022-05-25:
Hi,
1. Horizon's main window has a menu button (on the right side) that allows me to create a new window and with this open a 2nd project. This way I can have 2 projects side-by-side and compare schematics, board, whatever. This is very nice. However, closing one main window closes the other as well. This is probably because all open windows (projects) are part of one process.
Can we have additional logic that, if there is more than one main window, only the window is closed where the Quit menu command was issued? (like with the schematic or board window...)

2. The main window has 2 list boxes, one for the pools and one for the projects. If a pool is selected the main window changes the client area which becomes the Pool Manager. On the left side a button with the "<" character appears, so we can go back to the main client area (with the 2 list boxes). While the client area with the Pool Manager is visible the menu button (on the right side) that allows me to open a new window, etc. is still accessible which confuses the heck out of me, probably because the schematic and board windows have the same menu and selecting "Quit" only closes the window, but in the Pool Manager the whole app is shut down. I understand why it happens (from a programming point of view). 
Now, finally to my request/idea: while in the Pool Manager, can the menu button (on the right side) be invisible, or being replaced with a different button (same icon) where the "Quit" function has the same functionality as the "<" button? (BTW: the same should happen when the client area of the main window changes to the Project window...)

Sorry for the long text...

carrotIndustries, 2022-05-25:
My takeaway from this:

 - Remove the "Quit" menu item in the editors as it's confusing and redundant as there's the close button right next to it that does the same thing
 - Rename the Quit menu item in the pool/project manager to "Close all Windows" to make it more clear what it'll do.

How about that?

SonoQuium, 2022-05-25:
The 2 points you mention would work good for my 2nd point.
The 1st point I mentioned would still not be addressed: having several project main windows -> closing one will close all.
Maybe I misunderstood your 1st bullet...

carrotIndustries, 2022-05-25:
> having several project main windows -> closing one will close all.

Closing one project window only closes it and all associated editors. Using the quit/close all menu item will close all windows of the application.

SonoQuium, 2022-05-25:
Yes, that's what I meant.
Maybe changing the menu item from "Quit" to "Close all Windows and Quit" would help.
(or similar)
Thank you for creating this fine EDA package.
These few days testing it was fun, and I didn't find anything that would throw me off using it.
On KiCAD I just need 5 minutes... and I hate the UI, interaction and slowness.
(personal preference I guess)
Although, the new rounded traces that came with KiCAD 6 are nice for RF and Audio PCBs...
Thanks again.

carrotIndustries, 2022-05-25:
> Although, the new rounded traces that came with KiCAD 6 are nice for RF and Audio PCBs...

Good news, that'll land within the next weeks.

commit 02783550 (2022-05-25): clean up primary menus

imp: remove quit as it's redundant

pool/project manager: rename quit to make it more clear that it refers
to all windows rather than just the current one

Also, the GNOME HIG recommends not having these menu items: https://developer.gnome.org/hig/patterns/controls/menus.html

closes #667
closed 2022-05-25 by carrotIndustries