horizon-eda/horizon #674: Schematic: exception when placing part

moshee, 2022-06-26:
I added the Panasonic resistors pool to my project. Then I tried to place a part from that pool and got this exception.

[image]

The part chooser window was already open when I added the pool.

carrotIndustries, 2022-06-26:
Hm, works for me, seems like we need some more detailed steps to reproduce this. Does it work after reopening the project or updating the project pool.

moshee, 2022-06-28:
I was able to reproduce the issue. Here is what I did step by step:

* Started with schematic, part picker, pool manager open.
* Opened the pools window from the pool manager and located and installed the Yageo Chip Capacitors pool.
* Closed the pools window
* Closed the pool manager and reopened it from the start window
* Opened the settings tab, selected the new pool, added it, hit save, then hit refresh.
* Switched to the part picker, selected a capacitor using the parametric filter, and double click.
* Get the error.

What seems to be the problem is that it's looking in my local project folder for the parts, but they are not there - they have been installed in the default pools folder in ~\Documents.

Interestingly, as I switched back to the schematic and right clicked to exit the place part tool, it crashed the schematic editor. After reopening, the place part operation worked as expected.

carrotIndustries, 2022-06-28:
I see, the problem is that you installed the pool while the schematic editor was open and thus didn't know about the newly-instaleld pool as it only reads the list of installed pools on startup.

The solution is that the pool/project manager will tell the schematic editor to reload the list of installed pools after adding or removing one.

commit c4bfb893 (2022-06-29): imp: reload list of installed pools if it changed

fixes #674

changelog: Bugfixes/Schematic Editor: properly handle pools that are
installed while the editor is running
closed 2022-06-29 by carrotIndustries