horizon-eda/horizon #677: Crash when download (core) pool

oxplot, 2022-07-23:
Going into pool management screen and downloading the core pool results in a crash soon after the repository cloning starts:

[image]

The only output on the terminal is:

```sh
Gdk-Message: 01:51:48.623: Error flushing display: Resource temporarily unavailable
```

I'm running `2.3.1 Iceblink` on Arch Linux on Gnome with Wayland. Searching around the web seems to suggest that wayland is to blame but maybe apps can do a workaround if that's the case.

carrotIndustries, 2022-07-23:
I can't reproduce this on weston. Can you provide a backtrace?

oxplot, 2022-07-23:
I ran it with gdb on a non-debug build just to see if there was anything useful (before re-building the whole thing). Looks like the app handles the error and exits:

```
[Thread 0x7fffe19ff640 (LWP 316350) exited]
Gdk-Message: 02:45:57.463: Error flushing display: Resource temporarily unavailable
[Thread 0x7fffe11fe640 (LWP 316342) exited]
[Thread 0x7fffe37fe640 (LWP 316278) exited]
[Thread 0x7fffe3fff640 (LWP 316277) exited]
[Thread 0x7ffff1475640 (LWP 316275) exited]
[Thread 0x7ffff24b6640 (LWP 316273) exited]
[Thread 0x7ffff4322c00 (LWP 315907) exited]
[Thread 0x7ffff0c31640 (LWP 316276) exited]
[New process 315907]
[Inferior 1 (process 315907) exited with code 01]
(gdb) backtrace
No stack.
(gdb)
```

Do you have any instructions on how to further debug this?

carrotIndustries, 2022-07-28:
You might try setting a breakpoint on `exit`. If that doesn't work, rebuild gtk with debug symbols and set the breakpoint there: https://gitlab.gnome.org/GNOME/gtk/-/blob/gtk-3-24/gdk/wayland/gdkeventsource.c#L67

oxplot, 2022-07-28:
Thanks. Here's a backtrace. Had to run the download a few times to hit it:

```
Thread 1 "horizon-eda" hit Breakpoint 1, gdk_event_source_prepare (base=base@entry=0x101348e50, timeout=timeout@entry=0x7fffffffd984) at ../gtk/gdk/wayland/gdkeventsource.c:67
67            g_message ("Error flushing display: %s", g_strerror (errno));
(gdb) bt -full
#0  gdk_event_source_prepare (base=base@entry=0x101348e50, timeout=timeout@entry=0x7fffffffd984)
    at ../gtk/gdk/wayland/gdkeventsource.c:67
        source = 0x101348e50
        display = 0x101304050
        __func__ = "gdk_event_source_prepare"
#1  0x00007ffff67f583b in g_main_context_prepare (context=0x101311f20, priority=0x7fffffffda48) at ../glib/glib/gmain.c:3737
        begin_time_nsec = 15552474429618
        result = <optimized out>
        prepare = 0x7ffff6b6a380 <gdk_event_source_prepare>
        source_timeout = -1
        i = <optimized out>
        n_ready = 0
        current_priority = 2147483647
        source = 0x101348e50
        iter = {context = 0x101311f20, may_modify = 1, current_list = 0x7fffe4008e60 = {0x7fffe4008e40}, source = 0x101348e50}
#2  0x00007ffff684bdc6 in g_main_context_iterate.constprop.0
    (context=context@entry=0x101311f20, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>)
    at ../glib/glib/gmain.c:4191
        max_priority = 2147483647
        timeout = 16
        some_ready = <optimized out>
        nfds = <optimized out>
        allocated_nfds = 5
        fds = <optimized out>
        begin_time_nsec = 15552474429467
#3  0x00007ffff67f3392 in g_main_context_iteration (context=context@entry=0x101311f20, may_block=may_block@entry=1)
    at ../glib/glib/gmain.c:4276
        retval = <optimized out>
#4  0x00007ffff6a0f30e in g_application_run (application=0x1012b2210, argc=<optimized out>, argv=0x7fffffffdc58)
    at ../glib/gio/gapplication.c:2569
        arguments = 0x1013e3980
        status = 0
        context = 0x101311f20
        acquired_context = <optimized out>
        __func__ = "g_application_run"
#5  0x00000001002836d7 in  ()
#6  0x00007ffff6029290 in __libc_start_call_main
     (main=main@entry=0x100283670, argc=argc@entry=1, argv=argv@entry=0x7fffffffdc58) at ../sysdeps/nptl/libc_start_call_main.h:58
        self = <optimized out>
        result = <optimized out>

                      unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140737488346200, 2140561145519421033, 0, 140737488346216, 4313334208, 140737354125312, -2140561144332274071, -2140580912151513495}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x7fffffffdc58, 0x1}, data = {prev = 0x0, cleanup = 0x0, canceltype = -9128}}}
        not_first_call = <optimized out>
#7  0x00007ffff602934a in __libc_start_main_impl
     (main=0x100283670, argc=1, argv=0x7fffffffdc58, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdc48) at ../csu/libc-start.c:389
#8  0x000000010029e565 in  ()
(gdb)
```

oxplot, 2022-07-28:
Also, this may be relevant: https://gitlab.gnome.org/GNOME/gnome-shell/-/issues/3405#note_964698

carrotIndustries, 2022-07-28:
Hm, that's less helpful than I was hoping for. Can you run `inf thr` to check what the other threads are doing? My suspicion is that one of the background threads might be calling into Gtk even though it shouldn't.

oxplot, 2022-07-28:
Here it goes:

```
(gdb) inf thr
  Id   Target Id                                       Frame
* 1    Thread 0x7ffff4311c00 (LWP 3540) "horizon-eda"  gdk_event_source_prepare (base=base@entry=0x1013483a0,
    timeout=timeout@entry=0x7fffffffd984) at ../gtk/gdk/wayland/gdkeventsource.c:67
  2    Thread 0x7ffff24b6640 (LWP 3553) "gmain"        0x00007ffff6105c3f in __GI___poll (fds=0x10130b240, nfds=2,
    timeout=3996) at ../sysdeps/unix/sysv/linux/poll.c:29
  4    Thread 0x7ffff1475640 (LWP 3555) "gdbus"        0x00007ffff6105c3f in __GI___poll (fds=0x1013a0a40, nfds=7, timeout=-1)
    at ../sysdeps/unix/sysv/linux/poll.c:29
  5    Thread 0x7ffff0c31640 (LWP 3556) "dconf worker" 0x00007ffff6105c3f in __GI___poll (fds=0x1013455b0, nfds=1, timeout=-1)
    at ../sysdeps/unix/sysv/linux/poll.c:29
  6    Thread 0x7fffe3fff640 (LWP 3557) "ZMQbg/Reaper" 0x00007ffff6111c9e in epoll_wait (epfd=17, events=0x7fffe3ffdca0,
    maxevents=256, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  7    Thread 0x7fffe37fe640 (LWP 3558) "ZMQbg/IO/0"   0x00007ffff6111c9e in epoll_wait (epfd=19, events=0x7fffe37fcca0,
    maxevents=256, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
  25   Thread 0x7fffe11fe640 (LWP 3631) "horizon-eda"  0x00007ffff5a20f3e in ?? () from /usr/lib/libz.so.1
```

oxplot, 2022-07-28:
And backtrace for all threads:

```
(gdb) thread apply all bt -full

Thread 25 (Thread 0x7fffe11fe640 (LWP 3631) "horizon-eda"):
#0  0x00007ffff5a20f3e in  () at /usr/lib/libz.so.1
#1  0x00007ffff5a2aeab in inflate () at /usr/lib/libz.so.1
#2  0x00007ffff64ce3e5 in git_zstream_get_output_chunk (out=<optimized out>, out_len=0x7fffe11fb538, zstream=0x7fffc81166c0) at /usr/src/debug/libgit2-1.4.4/src/zstream.c:123
        in_used = <optimized out>
#3  0x00007ffff646b858 in git_packfile_stream_read (obj=0x7fffc81166b0, buffer=0x7fffc8117570, len=<optimized out>) at /usr/src/debug/libgit2-1.4.4/src/zstream.c:72
        window_len = 1609
        in = <optimized out>
        error = 0
#4  0x00007ffff644bd40 in read_object_stream (stream=<optimized out>, idx=<optimized out>) at /usr/src/debug/libgit2-1.4.4/src/indexer.c:295
        read = <optimized out>
        stream = 0x7fffc81166b0
        entry_start = 23427517
        entry_size = 663
        type = GIT_OBJECT_OFS_DELTA
        w = 0x0
        error = <optimized out>
        error = <optimized out>
        mwf = <optimized out>
#5  read_stream_object (stats=0x7fffc815b6e8, idx=0x7fffc8116660) at /usr/src/debug/libgit2-1.4.4/src/indexer.c:776
        stream = 0x7fffc81166b0
        entry_start = 23427517
        entry_size = 663
        type = GIT_OBJECT_OFS_DELTA
        w = 0x0
        error = <optimized out>
        error = <optimized out>
        mwf = <optimized out>
#6  git_indexer_append (idx=<optimized out>, data=<optimized out>, size=<optimized out>, stats=<optimized out>) at /usr/src/debug/libgit2-1.4.4/src/indexer.c:878
        error = <optimized out>
        mwf = <optimized out>
#7  0x00007ffff64bec63 in git_smart__download_pack (transport=0x7fffc81d2720, repo=<optimized out>, stats=0x7fffc815b6e8) at /usr/src/debug/libgit2-1.4.4/src/transports/smart_protocol.c:599
        pkt = 0x7fffc818ee60
        t = 0x7fffc81d2720
        buf = 0x7fffc81d2908
        odb = 0x7fffc812f3e0
        writepack = 0x7fffc810f380
        error = 0
        npp = {callback = 0x100861900, payload = 0x102269350, stats = 0x7fffc815b6e8, last_fired_bytes = 23424868}
        progress_cb = 0x100861900
        progress_payload = <optimized out>
#8  0x00007ffff648a8bd in git_fetch_download_pack (remote=0x7fffc815b5f0) at /usr/src/debug/libgit2-1.4.4/src/fetch.c:199
        t = <optimized out>
        to_active = <optimized out>
        specs = {_alloc_size = 0, _cmp = 0x0, contents = 0x0, length = 0, flags = 1}
        refs = {_alloc_size = 0, _cmp = 0x7ffff647dd30 <remote_head_cmp>, contents = 0x0, length = 0, flags = 0}
        i = <optimized out>
        error = <optimized out>
#9  git_remote__download (remote=remote@entry=0x7fffc815b5f0, refspecs=refspecs@entry=0x0, opts=opts@entry=0x7fffe11fbb20) at /usr/src/debug/libgit2-1.4.4/src/remote.c:1309
        to_active = <optimized out>
        specs = {_alloc_size = 0, _cmp = 0x0, contents = 0x0, length = 0, flags = 1}
        refs = {_alloc_size = 0, _cmp = 0x7ffff647dd30 <remote_head_cmp>, contents = 0x0, length = 0, flags = 0}
        i = <optimized out>
        error = <optimized out>
#10 0x00007ffff648ae6a in git_remote_fetch (remote=0x7fffc815b5f0, refspecs=0x0, opts=0x7fffe11fbb20, reflog_message=0x7fffc8187eb0 "clone: from https://github.com/horizon-eda/horizon-pool.git") at /usr/src/debug/libgit2-1.4.4/src/remote.c:1373
        error = 0
        update_fetchhead = 0
        tagopt = GIT_REMOTE_DOWNLOAD_TAGS_ALL
--Type <RET> for more, q to quit, c to continue without paging--
        prune = false
        reflog_msg_buf = {ptr = 0x7ffff650ed50 <git_str.initstr> "", asize = 0, size = 0}
        connect_opts = {version = 1, callbacks = {version = 1, sideband_progress = 0x0, completion = 0x0, credentials = 0x0, certificate_check = 0x0, transfer_progress = 0x100861900, update_tips = 0x0, pack_progress = 0x0, push_transfer_progress = 0x0, push_update_reference = 0x0, push_negotiation = 0x0, transport = 0x0, remote_ready = 0x0, payload = 0x102269350, resolve_url = 0x0}, proxy_opts = {version = 1, type = GIT_PROXY_NONE, url = 0x0, credentials = 0x0, certificate_check = 0x0, payload = 0x0}, follow_redirects = GIT_REMOTE_REDIRECT_INITIAL, custom_headers = {strings = 0x0, count = 0}}
#11 0x00007ffff641796a in clone_into (branch=<optimized out>, co_opts=0x7fffe11fbbf8, opts=0x7fffe11fbc88, _remote=0x7fffc819a6e0, repo=0x7fffc80fa8c0) at /usr/src/debug/libgit2-1.4.4/src/clone.c:415
        remote = 0x7fffc815b5f0
        error = <optimized out>
[... 183 lines cut ...]
        max_priority = 2147483647
        timeout = 16
        some_ready = <optimized out>
        nfds = <optimized out>
        allocated_nfds = 5
        fds = <optimized out>
        begin_time_nsec = 257102421153
#3  0x00007ffff67f3392 in g_main_context_iteration (context=context@entry=0x101311f20, may_block=may_block@entry=1) at ../glib/glib/gmain.c:4276
        retval = <optimized out>
#4  0x00007ffff6a0f30e in g_application_run (application=0x1012b2210, argc=<optimized out>, argv=0x7fffffffdc58) at ../glib/gio/gapplication.c:2569
        arguments = 0x1012c8c10
        status = 0
        context = 0x101311f20
        acquired_context = <optimized out>
        __func__ = "g_application_run"
#5  0x00000001002836d7 in  ()
#6  0x00007ffff6029290 in __libc_start_call_main (main=main@entry=0x100283670, argc=argc@entry=1, argv=argv@entry=0x7fffffffdc58) at ../sysdeps/nptl/libc_start_call_main.h:58
        self = <optimized out>
[... cut ...]
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140737488346200, -7456370708868060423, 0, 140737488346216, 4313334208, 140737354125312, 7456370707915123449, 7456353899255999225}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x7fffffffdc58, 0x1}, data = {prev = 0x0, cleanup = 0x0, canceltype = -9128}}}
        not_first_call = <optimized out>
#7  0x00007ffff602934a in __libc_start_main_impl (main=0x100283670, argc=1, argv=0x7fffffffdc58, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdc48) at ../csu/libc-start.c:389
#8  0x000000010029e565 in  ()
```

carrotIndustries, 2022-07-30:
Seems like the background thread is well-behaved and doing what it's supposed to to. One thing that might be an issue is that the frequent updates from the background thread might be messing up the compositor in some way or another.

You might try commenting out https://github.com/horizon-eda/horizon/blob/master/src/pool-prj-mgr/pools_window/pool_download_window.cpp#L126 and https://github.com/horizon-eda/horizon/blob/master/src/pool-prj-mgr/pools_window/pool_download_window.cpp#L138 and check if that makes a difference.

oxplot, 2022-08-03:
> You might try commenting out

Tried with same result. Still dies with the same error message:

```
(gdb) thread apply all bt -full

Thread 23 (Thread 0x7fffe11fe640 (LWP 121707) "horizon-eda"):
#0  __GI___ioctl (fd=42, request=request@entry=1074041865) at ../sysdeps/unix/sysv/linux/ioctl.c:36
        args = {{gp_offset = 16, fp_offset = 0, overflow_arg_area = 0x7fffe11fccb0, reg_save_area = 0x7fffe11fcc70}}
        r = 0
#1  0x00007ffff6993cba in btrfs_reflink_with_progress (cancellable=0x0, error=0x7fffe11fcd50, progress_callback_data=0x0, progress_callback=0x0, info=0x101b6d360, out=0x7fffc993e990, in=0x101d20db0) at ../glib/gio/gfile.c:3162
        source_size = <optimized out>
        ret = <optimized out>
        fd_in = 41
        fd_out = <optimized out>
        errsv = <optimized out>
        reflink_err = 0x0
        out = 0x7fffc993e990
        attrs_to_read = <optimized out>
        do_set_attributes = 1
        ret = 0
        info = <optimized out>
        tmp_error = 0x0
        file_in = 0x101d20db0
        in = 0x101d20db0
        target = <optimized out>
        create_flags = <optimized out>
        iface = <optimized out>
        my_error = 0x7fffc80dab00
        res = <optimized out>
        __func__ = "g_file_copy"
#2  file_copy_fallback (error=0x7fffe11fce50, progress_callback_data=0x0, progress_callback=0x0, cancellable=0x0, flags=<optimized out>, destination=0x1014d33a0, source=0x1014d4480) at ../glib/gio/gfile.c:3354
        reflink_err = 0x0
        out = 0x7fffc993e990
        attrs_to_read = <optimized out>
        do_set_attributes = 1
        ret = 0
        info = <optimized out>
        tmp_error = 0x0
        file_in = 0x101d20db0
        in = 0x101d20db0
        target = <optimized out>
        create_flags = <optimized out>
        iface = <optimized out>
        my_error = 0x7fffc80dab00
        res = <optimized out>
        __func__ = "g_file_copy"
#3  g_file_copy (source=0x1014d4480, destination=0x1014d33a0, flags=<optimized out>, cancellable=0x0, progress_callback=0x0, progress_callback_data=0x0, error=0x7fffe11fce50) at ../glib/gio/gfile.c:3566
        iface = <optimized out>
        my_error = 0x7fffc80dab00
        res = <optimized out>
        __func__ = "g_file_copy"
#4  0x00007ffff7535ca8 in Gio::File::copy(Glib::RefPtr<Gio::File> const&, Gio::FileCopyFlags) (this=<optimized out>, destination=<optimized out>, flags=<optimized out>) at gio/giomm/file.cc:639
        gerror = 0x0
        res = <optimized out>
#5  0x0000000100863330 in  ()
#6  0x000000010085f0f7 in  ()
#7  0x00007ffff7cd6183 in std::execute_native_thread_routine(void*) (__p=0x102100f20) at /usr/src/debug/gcc/libstdc++-v3/src/c++11/thread.cc:82
#8  0x00007ffff608c54d in start_thread (arg=<optimized out>) at pthread_create.c:442
        ret = <optimized out>
        pd = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140736970352192, 4980239827218458383, 140737488340094, 0, 140737488340095, 140736961961984, -4980242852008433905, -4980222881279383793}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
#9  0x00007ffff6111874 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:100

Thread 7 (Thread 0x7fffe37fe640 (LWP 121664) "ZMQbg/IO/0"):
#0  0x00007ffff6111c9e in epoll_wait (epfd=19, events=0x7fffe37fcca0, maxevents=256, timeout=-1) at ../sysdeps/unix/sysv/linux/epoll_wait.c:30
        sc_ret = -4
        sc_cancel_oldtype = 0
[... 146 lines cut ...]
#2  0x00007ffff684bdc6 in g_main_context_iterate.constprop.0 (context=context@entry=0x101311f20, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../glib/glib/gmain.c:4191
        max_priority = 2147483647
        timeout = 16
        some_ready = <optimized out>
        nfds = <optimized out>
        allocated_nfds = 5
        fds = <optimized out>
        begin_time_nsec = 177665345214414
#3  0x00007ffff67f3392 in g_main_context_iteration (context=context@entry=0x101311f20, may_block=may_block@entry=1) at ../glib/glib/gmain.c:4276
        retval = <optimized out>
#4  0x00007ffff6a0f30e in g_application_run (application=0x1012b2210, argc=<optimized out>, argv=0x7fffffffdc58) at ../glib/gio/gapplication.c:2569
        arguments = 0x10144e9e0
        status = 0
        context = 0x101311f20
        acquired_context = <optimized out>
        __func__ = "g_application_run"
#5  0x0000000100283697 in  ()
#6  0x00007ffff6029290 in __libc_start_call_main (main=main@entry=0x100283630, argc=argc@entry=1, argv=argv@entry=0x7fffffffdc58) at ../sysdeps/nptl/libc_start_call_main.h:58
        self = <optimized out>
        result = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140737488346200, 4980239827218327311, 0, 140737488346216, 4313334208, 140737354125312, -4980239828412274929, -4980222827088845041}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x7fffffffdc58, 0x1}, data = {prev = 0x0, cleanup = 0x0, canceltype = -9128}}}
        not_first_call = <optimized out>
#7  0x00007ffff602934a in __libc_start_main_impl (main=0x100283630, argc=1, argv=0x7fffffffdc58, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdc48) at ../csu/libc-start.c:389
#8  0x000000010029e525 in  ()
```

unrelentingtech, 2022-08-14:
Just encountered that as I wanted to redownload the pool.

Looks like it's this GTK bug: https://gitlab.gnome.org/GNOME/gtk/-/issues/124

carrotIndustries, 2022-08-14:
Wondering why it only happens on pool download. There are other places that frequently update the UI such as pool update and checks.