horizon-eda/horizon #681: STEP export fails with zero length polygon segments

tkornack, 2022-08-01:
It's not hard to inadvertently create a zero length polygon segment (click twice in the same spot), and generally it seems quite harmless:

[image]

However, the STEP exporter fails with 

```
Board outline…
Board cutouts…
Holes…
Creating board…
Error: 0xffffaddaa3a0 : StdFail_NotDone: BRep_API: command not done

```

It's easy enough to fix once you know what's wrong, but I wonder if you could consider one of these:
- Check for zero length segments (either in Rules or in the STEP exporter)
- Automatically remove zero length segments

carrotIndustries, 2022-08-01:
The recent commits on master address this in multiple ways:

 - The draw polygon tool doesn't create zero-length edges anymore when clicking twice on the same point
 - The STEP export removes zero-length edges
 - The preflight check checks for zero-length edges and flags them as a warning

closed 2022-08-01 by carrotIndustries

tkornack, 2022-08-01:
That's great!

commit b36d297f (2022-08-01): step export: don't crash on zero-length edges in outline polygons

closes #681

changelog: Bugfixes/Board Editor: STEP export: don't crash on
zero-length edges in outline polygons